STMicroelectronics/STM32CubeF4 #90: MX_DMA_Init called after MX_DFSDM causes the DMA interrupt to not work

noomio, 2021-09-09:
NUCLEO-F413ZH
STM32CubeIDE

STM32CubeMx package STM32F4 series: 1.26.2

Configuring the DFSDM2 with DMA didnt fire the DMA2 interrupt and DMA2_Stream0_IRQHandler was never called.

The code generated with CubeMX:

`
int main(void)
{
  /* USER CODE BEGIN 1 */

  /* USER CODE END 1 */

  /* MCU Configuration--------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* USER CODE BEGIN Init */

  /* USER CODE END Init */

  /* Configure the system clock */
  SystemClock_Config();

/* Configure the peripherals common clocks */
  PeriphCommonClock_Config();

  /* USER CODE BEGIN SysInit */

  /* USER CODE END SysInit */

  /* Initialize all configured peripherals */
  MX_GPIO_Init();
  MX_ADC1_Init();
  MX_DFSDM2_Init();
  MX_I2S2_Init();
  MX_USB_OTG_FS_PCD_Init();
  MX_SAI1_Init();
  MX_UART4_Init();
  MX_TIM6_Init();
  MX_DMA_Init();
  MX_USART2_UART_Init();
`

Fix is to call MX_DMA_Init(); first and then all the rest.

RKOUSTM, 2021-09-20:
Hi @noomio and @DboubaMefteh,

Thank you for your contribution. You are absolutely right about this point. Actually, the point you raised has already been dealt and fixed internally. The fix will be made available in the frame of a future release of CubeMX. 

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to our ecosystem (namely the Cube MX tool), please allow me to close it.

Thank you for your contribution. We are looking forward to reading from you again.

With regards,

RKOUSTM, 2021-09-20:
ST Internal Reference: 111339

closed 2021-09-20 by RKOUSTM