STMicroelectronics/STM32CubeF4 #91: Missing struct member init in BSP LIS3DSH driver

kevinpt, 2021-09-14:
There is a call to LIS3DSH_InterruptConfig() in lis3dsh.c in the BSP components that is missing an init value for State_Machine1_Interrupt. This is flagged for me by GCC with -Werror=uninitialized. I don't know what an appropriate init value should be for this but it should compile cleanly nonetheless.

https://github.com/STMicroelectronics/STM32CubeF4/blob/4aba24d78fef03d797a82b258f37dbc84728bbb5/Drivers/BSP/Components/lis3dsh/lis3dsh.c#L222

RKOUSTM, 2021-09-20:
Hi @kevinpt,

Thank you for your report. It has been forwarded to our development teams. I will get to you as soon as they provide their feedback.

With regards,

RKOUSTM, 2021-09-21:
Hi @kevinpt,

Thank you for your contribution. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-09-21:
ST Internal Reference: 114019

RKOUSTM, 2022-03-11:
Hi @kevinpt, 

Thank you for your contribution. This issue has been fixed in the frame of version v1.27.0 of the STM32CubeF4 published recently. Please allow me then to close this thread.

Thank you again for your contribution.

With regards,

closed 2022-03-11 by RKOUSTM