STMicroelectronics/STM32CubeF4 #93: GCC linker scripts have an overly restrictive license

kevinpt, 2021-09-19:
I noticed that all the GCC linker scripts have this license notice:

```
**  (c)Copyright Ac6.
**  You may use this file as-is or modify it according to the needs of your
**  project. Distribution of this file (unmodified or modified) is not
**  permitted. Ac6 permit registered System Workbench for MCU users the
**  rights to distribute the assembled, compiled & linked contents of this
**  file as part of an application binary file, provided that it is built
**  using the System Workbench for MCU toolchain.
```

It pretty clearly states that nobody can distribute linked binaries without using System Workbench. I think scripts with a more permissive license are in order if this is going to truly be an open source project.

Presumably this also affects the repos for the other device families.

An example script with the license header:

https://github.com/STMicroelectronics/STM32CubeF4/blob/4aba24d78fef03d797a82b258f37dbc84728bbb5/Projects/STM32F429I-Discovery/Templates/SW4STM32/STM32F429I_DISCO/STM32F429ZITx_FLASH.ld#L21

RKOUSTM, 2021-09-27:
Hi @kevinpt,

Thank you for your contribution. The reported point will be forwarded to our development teams. I will get to you as soon as they provide their feedback.

Thank you again for your contribution and for your patience.

With regards,

RKOUSTM, 2022-02-28:
ST Internal Reference: 123572

ALABSTM, 2023-04-10:
Hi @kevinpt,

I hope you are fine. This point you reported should be no more an issue once all projects of this firmware would be **migrated** to `CubeIDE` format.

With regards,

TOUNSTM, 2023-11-23:
Hello @kevinpt,

Thank you for this report. As @ALABSTM said, in the latest version of **STM32CubeF4 v1.28.0** you'll only find STM32CubeIDE projects. Please allow me to close this thread.

With Regards,

closed 2023-11-23 by TOUNSTM