STMicroelectronics/STM32CubeF4 #94: Nucleo F411RE SPI not working

xRadx, 2021-10-07:
**Describe the set-up**
 * Nucleo F411RE
 * STM32IDE 1.7.0, F4 package 1.26.0

**Describe the bug**
SPI does not work for the board, but works on F446RE with the same CUBEMX SPI settings and same instructions/commands inside of main.c 

**How To Reproduce**
1. Indicate the global behavior of your application project: Sending some random data via SPI1 od SPI2 is not visible for another MCU.

SPI settings:
Motorolla, 8 bits, MSB first, 2MB/s, polarity LOW, 1 edge, CRC disabled, NSS Software.
PB1 set to output as CS 

main.c:
HAL_GPIO_WritePin(GPIOB, GPIO_PIN_1, 1);
uint8_t tx[4],
  tx[0]=65;
  tx[1]=66;
  tx[2]=67;
  tx[3]=68;
  HAL_Delay(3000);
while(1)
{
	  HAL_GPIO_WritePin(GPIOB, GPIO_PIN_1, 0);
	  HAL_SPI_Transmit(&hspi2,  &tx, 4, 10);
	  HAL_GPIO_WritePin(GPIOB, GPIO_PIN_1, 1);
	  HAL_Delay(100);
  }

ASELSTM, 2021-10-20:
Hi @xRadx,

Would you please share the IOC file in order to allow a better analysis of the problem.

With regards,

xRadx, 2021-11-08:
[SPI_Test.zip](https://github.com/STMicroelectronics/STM32CubeF4/files/7495057/SPI_Test.zip)

Zipped ioc file attached.

xRadx, 2021-11-08:
[SPI_Test.zip](https://github.com/STMicroelectronics/STM32CubeF4/files/7495099/SPI_Test.zip)

ioc attached

ASELSTM, 2023-07-11:
Hi @xRadx,

First, please excuse this late answer. 

Based on the IOC file you have share with us, we have generated the project using the latest version of the SMT32CubeMX and run it on an F411RE Nucleo board. We did not detected any issue with the SPI, it is working perfectly as expected. 

Find below the data frame recorded on the MOSI line. We have made few update on the data to be transferred by setting the tx buffer to `tx[] = "SPI2"`.

[image]

Please make sure you are correctly wiring the transmitter and receive boards. Otherwise, I recommend you to use the latest version of the [STM32CubeMx](https://www.st.com/en/development-tools/stm32cubemx.html) **v6.9.0**.

As there is no issue, please allow me thus to close this thread. 

With regards,

closed 2023-07-11 by ASELSTM