STMicroelectronics/STM32CubeF4 #101: EEprom flash hal driver incorrect lock/unlock sequence

paulusjacobus, 2021-11-08:
**Describe the set-up**
 * Disco F4 board MB997D
 * IDE 1.4

**Describe the bug**
Flash command won't program a memory location in flash

**How To Reproduce**
1. Indicate the global behavior of your application project. HAL_FLASH_Program() generates a Page Error

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
STM32CubeF4/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash.c

3. The use case that generates the problem.--> Emulate an EEPROM in Flash as per AN3969

4. How we can reproduce the problem. Just use the ST example to emulate an eeprom.
file:///C:/Users/paulj/Downloads/an3969-eeprom-emulation-in-stm32f40xstm32f41x-microcontrollers-stmicroelectronics.pdf

**Additional context**
Lastest Driver 1.6.2
      (#) FLASH Memory IO Programming functions: 
           (++) **Lock and Unlock the FLASH interface using HAL_FLASH_Unlock() and 
                HAL_FLASH_Lock() functions**---> the other way around....works
           (++) Program functions: byte, half word, word and double word
           (++) There Two modes of programming :
            (+++) Polling mode using HAL_FLASH_Program() function
            (+++) Interrupt mode using HAL_FLASH_Program_IT() function

You need to unlock the flash, program, and lock the flash. The hal driver does it the other way around.
Next issue is that the lock/unlock uses a interrupt handle &pFlash by default. Just use HAL_Unlock() and HAL_Lock() and it works perfectly'

**Screenshots**
If applicable, add screenshots to help explain your problem.

ALABSTM, 2022-01-30:
Hi @paulusjacobus,

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

RKOUSTM, 2022-02-28:
Hi @paulusjacobus,

Thank you for your report. The problem seems to be related to migration to the CubeIDE. However, this public organization on GitHub is intended to promote co-development and issues posted here shall exclusively be related to the software published within the different repositories. 

You can refer to this [EEPROM_Emulation](https://github.com/STMicroelectronics/STM32CubeF4/tree/master/Projects/STM32469I_EVAL/Applications/EEPROM/EEPROM_Emulation) application on STM324x9I_EVAL for emulate the EEPROM mechanism. The IDE proposed by ST (IAR, MDK-ARM and SW4STM32) for the EEPROM_Emulation application on STM32F4 board all of them are working well.

For any other issue (e.g. related to documentation, to hardware, to ecosystem, to user application, etc.), you shall address your questions to the [ST Community](https://community.st.com/s/topiccatalog). There you shall look for the adequate topic and submit your question or remark.

Please allow me to close this issue now. Thank you for your comprehension and contribution.

With regards,

closed 2022-02-28 by RKOUSTM