STMicroelectronics/STM32CubeF4 #103: incorrect comment

kitoma, 2021-11-17:
A comment in stm32f4xx_hal_uart.c is incorrect; see
https://github.com/kitoma/STM32CubeF4/compare/master...kitoma-patch-1

(I'm not sure what all this contributor agreement stuff is about, too much text too read for a simple comment update/pull-request)

ALABSTM, 2021-12-21:
Hi @kitoma,

Thank you for this report. The CLA (Contributor License Agreement) is to be signed only once. Once done, your pull-requests will be no more closed by the bot. I hope the text does not discourage you to sign it. Here is the [link](https://cla.st.com/cla). Choose the **individual** CLA for simplicity. We are looking forward to receiving other pull-requests from you.

With regards,

ALABSTM, 2021-12-22:
Hi @kitoma,

Your point has been logged into our internal database. An update shall be made available in a future release.

With regards,

ASELSTM, 2023-04-06:
Hi @kitoma,

Thank you for your contribution. This issue has been fixed in the frame of version v1.27.0 of the STM32CubeF4. Please allow me then to close this thread.

With regards,

closed 2023-04-06 by ASELSTM