STMicroelectronics/STM32CubeF4 #105: SDMMC_CmdSetRelAddMmc is needed to correctly address MMCs

dpgeorge, 2021-12-08:
The `MMC_InitCard()` function calls `SDMMC_CmdSetRelAdd()` which attempts to set the MMC address to 0 instead of the given RCA value.  The RCA for an MMC should be set to 1 or greater.

The STM32CubeH7 MMC driver has a fix for this where it uses `SDMMC_CmdSetRelAddMmc()`:
```c
uint32_t SDMMC_CmdSetRelAddMmc(SDMMC_TypeDef *SDMMCx, uint16_t RCA)
{
  SDMMC_CmdInitTypeDef  sdmmc_cmdinit;
  uint32_t errorstate;

  /* Send CMD3 SD_CMD_SET_REL_ADDR */
  sdmmc_cmdinit.Argument         = ((uint32_t)RCA << 16U);
  sdmmc_cmdinit.CmdIndex         = SDMMC_CMD_SET_REL_ADDR;
  sdmmc_cmdinit.Response         = SDMMC_RESPONSE_SHORT;
  sdmmc_cmdinit.WaitForInterrupt = SDMMC_WAIT_NO;
  sdmmc_cmdinit.CPSM             = SDMMC_CPSM_ENABLE;
  (void)SDMMC_SendCommand(SDMMCx, &sdmmc_cmdinit);

  /* Check for error conditions */
  errorstate = SDMMC_GetCmdResp1(SDMMCx, SDMMC_CMD_SET_REL_ADDR, SDMMC_CMDTIMEOUT);

  return errorstate;
}
```
I think this function needs to be ported to the STM32CubeF4 HAL and used in `MMC_InitCard()`.

ASELSTM, 2022-01-06:
Hi @dpgeorge,

Thank you for your contribution and for the proposed fix. We are actually aware about the point you raised out and have fixed it internally. The fix  is as you have proposed, the `SDMMC_CmdSetRelAddMmc` has been ported to the STM32CubeF4.

 This fix will be made available in the frame of a future release. So stay tuned and thank you once more for your contribution and for your patience.

With regards

ASELSTM, 2022-01-06:
ST Internal Reference: 83264

ASELSTM, 2022-03-11:
Hi @dpgeorge,

This issue has been fixed in the frame of version v1.27.0 of the STM32CubeF4 published recently. Please allow me then to close this thread.

With regards,

closed 2022-03-11 by ASELSTM