STMicroelectronics/STM32CubeF4 #106: Use of ATOMIC_SET_BIT ATOMIC_CLEAR_BIT macros in F4 LL UART driver

pavel-a, 2021-12-11:
Questioning use of  ATOMIC_SET_BIT ATOMIC_CLEAR_BIT macros  in LL_USART_EnableDirectionRx,  LL_USART_SetTransferDirection and some others:
https://github.com/STMicroelectronics/STM32CubeF4/blob/4aba24d78fef03d797a82b258f37dbc84728bbb5/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_usart.h#L449

The ATOMIC_SET_BIT ATOMIC_CLEAR_BIT macros are based on ldrex/strex instructions: see
https://github.com/STMicroelectronics/STM32CubeF4/blob/4aba24d78fef03d797a82b258f37dbc84728bbb5/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h#L234  and below.

ARM documentation says that lderx/strex should be used only on "normal" memory.
Can this be a problem when used on UART registers? 

See also community discussion:
https://community.st.com/s/question/0D53W00001FQqXxSAL/stm32f4-purpose-of-the-usage-of-atomicsetbit-atomicclearbit-macros-in-the-low-level-drivers-related-to-uart-peripheral

ASELSTM, 2022-01-26:
Hi @pavel-a,

As mentioned in the release note, the atomic register access has been used to handle the UART concurrent register access in case of race condition between Tx and Rx transfers.

Indeed, a case of race condition has been detected during Tx and Rx transfer which induce to simultaneous access to the CR1 and CR3 registers. A protection is thus required on all CR1 and CR3 registers in the Transmission and Reception APIs in the Polling, IT and DMA modes hence the use of the atomic register access.  The use of STREX/LDREX has been tested and it is fully applicable on the UART registers.

With regards,

pavel-a, 2022-01-27:
@ASELSTM thank you for clarifying this issue

closed 2022-02-01 by ASELSTM