STMicroelectronics/STM32CubeF4 #107: RAM layout in auto-generated Keil projects for i.e. STM32F437VG does not reflect real RAM layout

theadib, 2021-12-29:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
most recent versions I use as of 2021-12-29
* STMCubeMX 6.4.0
* Keil 5.36

**Describe the bug**
The STM32F437VG has several RAM sections that can be attributed by the IDE tools
1) 3 blocks SRAM @2000.0000h: 112kB, 16kB, 64kB, in total 196kB
2) 1 block CCM RAM @1000.0000h, 64kB

The first 3 blocks are in consecutive order with no gaps in between.
For those 3 blocks the IDE has only 1(one) entry in the graphical representation.

this is the problem:
If I have a large memory block reserved as DMA buffer. 
And that buffer spans over the boundries of 1(one) of these blocks.
Then DMA will fail/stop.

in my case I used a large buffer as DMA buffer for the lwip stack
`uint32_t dmabuffer[90kByte/4];
`
But as there where other variables declared before that, the 90kByte block spanned from the first over to the second RAM block.
The DMA failed under this memory layout because the DMA has problems when accessing several RAM modules.

As the number of RAM blocks addresseable in the GUI is very limited an other method should be chosen to define MEMORY layout.
Keil has the ability to create special scatter linker files.
probably that should be the method to address this issue.

**How To Reproduce**
1. Indicate the global behavior of your application project.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).

3. The use case that generates the problem.

4. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

ALABSTM, 2022-01-30:
Hi @theadib,

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

RKOUSTM, 2022-03-04:
Hi @theadib,

Thank you for your report. 
Actually, the point you reported is not directly related to the firmware published in this repository and is out of the scope of STMicroelectronics' activity. They are rather treated at [ST Community](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) level. Please allow me to redirect you to address your request in [this](https://community.st.com/s/question/0D50X00009XkYmMSAV/stm32f407-2-ram-banks) discussion section where the same issue is addressed.

Thank you again for your report. We are looking forward to reading from you again.

With regards,

closed 2022-03-04 by RKOUSTM

theadib, 2022-03-04:
Hello RKOUSTM,

it is quite easy to ignore that issue.
Of course the TSM32F4 libs per se does not have this issue.

The issue is the generated project files by STM32CubeMX application.
So it is definitely NOT an STCommunity issue.

Please follow up this issue in a more professional manner. And assign that
issue to the STM32CubeMX team.

In the last time I am more and more disappointed by the way how ST treat
his customers.

In the hope the words are heard ....

Regards,

Adib.

RKOUSTM, 2022-03-08:
Hi @theadib,

I hope you are fine. The issue you reported has been logged into an internal bug tracker. The fix will be made available in the frame of a future CubeMX release.

However, as it is related to the ecosystem (namely the Cube MX tool) rather than the firmware published within this repository (CMSIS, HAL, BSP, etc.), please allow me to keep this thread closed. 

In case you have other CubeMX-related issues to share, please submit them in the [ST Community](https://community.st.com/s/question/0D53W00000JhsU4SAJ/stm32cubemx-ide-issue-throws-warning-when-using-macros) in the CubeMX related section. There you shall find people who will address your request and whom you will hopefully find a solution with.

Thank you for your comprehension and thank you again for having reported.

With regards,

RKOUSTM, 2022-03-08:
ST Internal Reference: 124011

theadib, 2022-03-08:
Merci. Adib.