STMicroelectronics/STM32CubeF4 #108: Disable (-Wunused-parameter) on F4 series

DanielHaubjergVestergaard, 2022-01-04:
Hi  STMicroelectronics /STM32CubeF4

stm32duino/Arduino_Core_STM32

When "Compiler warnings" is set to "ALL" there will be a warning on the F4 series.
This has been tested with Core version 2.2.0, with (Example Blink) and Arduino 1.8.19

Drivers / STM32F4xx_HAL_Driver / Inc / stm32f4xx_ll_adc.h: In function 'uint32_t LL_ADC_DMA_GetRegAddr (ADC_TypeDef *, uint32_t)':
Drivers / STM32F4xx_HAL_Driver / Inc / stm32f4xx_ll_adc.h: 1881: 76: warning: unused parameter 'Register' [-Wunused-parameter]

Drivers / STM32F4xx_HAL_Driver / Inc / stm32f4xx_ll_adc.h: In function 'uint32_t LL_ADC_IsActiveFlag_MST_EOCS (ADC_Common_TypeDef *)':
Drivers / STM32F4xx_HAL_Driver / Inc / stm32f4xx_ll_adc.h: 4396: 75: warning: unused parameter 'ADCxy_COMMON' [-Wunused-parameter]

[image]

RKOUSTM, 2022-01-07:
Hi @DanielHaubjergVestergaard,

Thank you for your contribution. The issue you pointed out has been confirmed. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you once more for your contribution.

With regards,

RKOUSTM, 2022-01-07:
ST Internal Reference: 120469

RKOUSTM, 2022-01-07:
ST Internal Reference: 111256

RKOUSTM, 2022-03-11:
Hi @DanielHaubjergVestergaard,

Thank you for your contribution. This issue has been fixed in the frame of version v1.27.0 of the STM32CubeF4 published recently. Please allow me then to close this thread.

Thank you again for your contribution.

With regards,

closed 2022-03-11 by RKOUSTM