STMicroelectronics/STM32CubeF4 #109: HAL_SRAM_Read_DMA and HAL_SRAM_Write_DMA use always weak functions

tguetle, 2022-01-12:
The functions `HAL_SRAM_Read_DMA` and `HAL_SRAM_Write_DMA` always set the DMA callback function pointers to the weak functions 
```
  hsram->hdma->XferCpltCallback  = HAL_SRAM_DMA_XferCpltCallback;
  hsram->hdma->XferErrorCallback = HAL_SRAM_DMA_XferErrorCallback;
```
and don't use the functions set with `HAL_SRAM_RegisterDmaCallback`

The function pointers `hsram->DmaXferCpltCallback` and `hsram->DmaXferErrorCallback` are never used.

A simple patch is to set the DMA callback function pointers in the functions `HAL_SRAM_Read_DMA` and `HAL_SRAM_Write_DMA` like this:
```
  hsram->hdma->XferCpltCallback  = hsram->DmaXferCpltCallback;
  hsram->hdma->XferErrorCallback = hsram->DmaXferErrorCallback;
```

ASELSTM, 2022-01-27:
Hi @tguetle,

Thank you for your contribution. We are actually aware about the point you raised out and have fixed it internally.

This fix will be made available in the frame of a future release. So stay tuned and thank you once more for your contribution and for your patience.

With regards

ASELSTM, 2022-03-11:
Hi @tguetle,

This issue has been fixed in the frame of version v1.27.0 of the STM32CubeF4 published recently. Please allow me then to close this thread.

With regards,

closed 2022-03-11 by ASELSTM