STMicroelectronics/STM32CubeF4 #113: HAL_Delay() infinite loop because systick not incrementing on stm32F4 boards due to bugs in system_stm32f4xx.c when making new default project running from flash

beachmiles, 2022-03-07:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.
stm32f469VI and stm32f479AI custom board
stm32cubeide IDE version 1.8.0 running on windows 10  and stm32cubemx 6.4.0

**Describe the bug**
A clear and concise description of what the bug is.
Creating a new stm32cubemx project for either of the stm32F4 chips using the stm32cube 1.26.0-1.26.2 drivers causes HAL_Delay to enter infinite loop because the systick timer to not increment and the default systick priority set to 15 instead of 0. Setting systick priority to 0 does not help.

**How To Reproduce**
1. Indicate the global behavior of your application project.
Create a new stm32cube project for the stm32f469VI or stm32f479AI (suspect same issue on all stm32F4 chips) using stm32cube 1.26.0-1.26.2 drivers.
Insert a HAL_Delay(5) in the while loop.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
system_stm32f4xx.c

3. The use case that generates the problem.
Trying to make a stm32cube project.

4. How we can reproduce the problem.
Make a new stm32cube project for the stm32f469VI or stm32f479AI (suspect same issue on all stm32F4 chips)

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.
Edit this section in system_stm32f4xx.c by doing the following 2 things. Full updated code snippet below
  1) uncommenting the "#define USER_VECT_TAB_ADDRESS" 
  2) adding the line #if defined(USER_VECT_TAB_ADDRESS)	

/* Note: Following vector table addresses must be defined in line with linker
         configuration. */
/*!< Uncomment the following line if you need to relocate the vector table
     anywhere in Flash or Sram, else the vector table is kept at the automatic
     remap of boot address selected */
#define USER_VECT_TAB_ADDRESS					//This needs to be uncommented out by default! -MILES

/*!< Uncomment the following line if you need to relocate your vector Table in
     Internal SRAM. */
/* #define VECT_TAB_SRAM */
#if defined(USER_VECT_TAB_ADDRESS)				// THIS LINE WAS PREVIOUSLY MISSING! -MILES

**Screenshots**
If applicable, add screenshots to help explain your problem.

fpistm, 2022-03-08:
Hi @beachmiles,

Are you flashing using Serial? (Thanks bootloader)?
If yes then this issue is probably the same than this one: https://github.com/stm32duino/Arduino_Core_STM32/pull/1653

By default the system_stm32f4xx.c template file does not restore the VTOR. In that case systick could not be incremented.
Moreover you should ensure clock config is properly reset to not use the PLL else if your system clock config try to reconfigure it then it will failed as PLL is already used.

beachmiles, 2022-03-08:
I am flashing / debugging using a stlink v3 usb programmer.
I just updated to stm32cubeide 1.9 with this version of the firmware STM32Cube FW_F4 V1.27.0

Creating a new project still had this line commented out #define USER_VECT_TAB_ADDRESS
which still keeps the systick from incrementing. Uncommenting it fixes the problem.
/* Note: Following vector table addresses must be defined in line with linker
         configuration. */
/*!< Uncomment the following line if you need to relocate the vector table
     anywhere in Flash or Sram, else the vector table is kept at the automatic
     remap of boot address selected */
#define USER_VECT_TAB_ADDRESS

vahidajalluian, 2022-03-15:
Dear @beachmiles 
Your solution worked also for me with F429.

RKOUSTM, 2022-04-08:
Hi @beachmiles,

Thank you your contribution. Actually, the point you raised has already been dealt and fixed internally. The fix will be made available in the frame of a future release.

Thank you once more for your contribution. We are looking forward to reading from you again.

With regards,

RKOUSTM, 2022-04-08:
ST Internal Reference: 116475

wilgaced, 2022-11-23:
> I am flashing / debugging using a stlink v3 usb programmer. I just updated to stm32cubeide 1.9 with this version of the firmware STM32Cube FW_F4 V1.27.0
>
> Creating a new project still had this line [...]

wilgaced, 2022-11-23:
It Work for STM32L496

beachmiles, 2023-01-18:
> Creating a new project still had this line commented out #define USER_VECT_TAB_ADDRESS which still keeps the systick from incrementing. Uncommenting it fixes the problem. /* Note: Following vector [...]

It looks like this is still "In Progress" as of 1/17/2023, it's unfortunate this is the case as I suspect many people across the world will have ripped many hairs out from this bug the past 10+ months.
https://github.com/orgs/STMicroelectronics/projects/2#card-79208657

beachmiles, 2023-02-17:
Verified this is still a bug in the stm32L496ZG-P on the Nucleo board.
I'm using stm32cubeide 1.11.2 using stm32cube firmware L4 V1.17.2
Can't beleive a major bug like this could persist for close to a year spanning at least the stm32F4 and stm32L4 lines.

beachmiles, 2023-02-23:
Dear @fpistm @ALABSTM , I verified this is still a bug when creating a new project for a stm32L4 (NUCLEO-L496ZG-P) with the original fix still working except the the 2nd step is no longer needed as the #if defined(USER_VECT_TAB_ADDRESS)	 line is now in both of the latest system_stm32f4xx.c and system_stm32l4xx.c files.
So now I only had to uncomment this line to get the system from freezing on initialization due to a frozen systick.
`#define USER_VECT_TAB_ADDRESS`

So it looks like ST needs to update these files with this one line fix unless there is a better/different fix in the works. Thanks!
https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c

https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/system_stm32l4xx.c

vahidajalluian, 2023-03-17:
I didn't like to say this like, out of morality, but as it has wasted a lot of time from developers who trusted developers of cube MicroXplorer, it is a pity that til now there had been no workaround for a line to be uncommented, and a line to be added.

stnigg, 2023-06-16:
Currently the lates version for L4 seems to be 1.17.2. The issue is present in this library at least for the STM32L431.
I must say it is disapointing to find out that this is known for this long and is still wasting my time.

fpistm, 2023-06-16:
> 

Hi @beachmiles 
I'm aware of this and fixed this in https://github.com/stm32duino/Arduino_Core_STM32. 
From my point of view, VTOR should be unconditionally set but I'm not working on Cube, I'm only use it for stm32duino core.
An internal bugtracker has been submitted and so I hope it will be resolved by concerned team.

mrpj100, 2023-06-21:
Can confirm this also affected me on an STM32L432,

timonburkard, 2023-06-29:
Issue also exists on STM32L433CB6 using STM32CubeIDE 1.12.1

Ashish-Jog, 2023-07-28:
Issue still exists on STM32H743IIT6 m CubeIDE Version :- 1.11.2

crnorris, 2023-11-17:
Issue exists with a fresh project built with STM32CubeIDE 1.13.2, STM32CubeMX 6.9.2, targeting STM32G030J6M6.

smelting, 2023-12-31:
Issue persistent with STM32G491RET6, STM32CubeIDE 1.11.0, STM32Cube FW_G4 V1.5.2. Wasted my time checking my board design/soldering because a blank project with a blink and delay was broken.

beachmiles, 2024-01-24:
What in the heck is the problem ST??? Close to a 2 year old open bug that can be fixed by 1 line!!! Cmon!!!

YocoZuna, 2024-05-23:
Issue persistent with STM32L051R8T6

DAflamingFOX, 2024-08-04:
Can confirm this issue is still persistent on STM32G431KBU6, STM32CubeIDE 1.16.0 & STM32Cube FW_G4 V1.6.0

The solution posted here still works...

It's been over 2 years, this is a joke right?

augustas534, 2024-09-02:
Issue still exists with a fresh project built with STM32CubeIDE 1.16.0, STM32Cube FW_U5 V1.6.0, targeting STM32U545RETxQ. 

Following fix does not work on this processor...

tbranyon, 2024-12-10:
This issue is present for STM32G473RCT with CubeMx 6.12.1 and with IDE 1.17.0. Wasted 2 days trying to track down an issue just to find that it's this Cube-generated bug, which experience has taught me I should have suspected from the beginning. Can't believe it's been known about and open for 2 years...You're killing me, ST.

ALABSTM, 2024-12-13:
Hi everyone,

Really sorry for any inconvenience caused.

Actually, in case the vector table is kept at its default location at the very beginning of the flash memory, **i.e.**, at address `0x0080'0000`, there is no need to initialize the VTOR register. Indeed, in this case, whether the VTOR is initialized with value `0x0080'0000` or left unchanged after reset (**i.e.**, with value `0x0000'0000`) makes no difference. Interrupts will be processed correctly.

On the other hand, any location other than the default one, macro `USER_VECT_TAB_ADDRESS` has to be defined, whether by uncommenting the corresponding line in the `system_stm32f4xx.c` file or at IDE level. Then, whether you define the `VECT_TAB_SRAM` macro the same way to place the table in RAM. Finally, you redefine macro `VECT_TAB_OFFSET` at your convenience.

**Question:** @beachmiles, @vahidajalluian, and all of you guys, would you mind precising where are you placing the vector table in the frame of your applications please? If it is at the default location, would you mind providing more details about how to reproduce the issue? From our side, the first tests made showed no issue.

Another aspect to consider is whether you are using a **bootloader** that might change the default value of the VTOR register.

With regards,

rasil1127, 2024-12-17:
Issue still exists with a fresh project built with STM32CubeIDE 1.17.0 targeting STM32L071RBT6.

uncommenting "#define USER_VECT_TAB_ADDRESS" does not fix the problem. 

Program stays in a loop of  `__weak uint32_t HAL_GetTick(void)
{
  return uwTick;
}`

vahidajalluian, 2024-12-19:
@ALABSTM Well I had no problem with the latest releases and it seems to me the problem has been solved since then by the CUBE code generator, I don't know in which version the problem was really solved. (Tested on STM32H743BIT6 and STM32F429BIT6).

ALABSTM, 2024-12-19:
Hi @rasil1127,

I generated a project using STM32CubeMX **6.12.0**, built it and ran it using EWARM **9.20.4**, and the application executed correctly. The `SysTick` interrupt is correctly processed, the `uwTick` variable is incremented at each call, and the `HAL_Delay()` function works fine.

Could you please ensure your implementation of the `SysTick_Handler()` function calls `HAL_IncTick()` , as shown below?

```c
void SysTick_Handler(void)
{
  HAL_IncTick();
}
```

With regards,

ALABSTM, 2024-12-19:
@rasil1127,

I gave it another try building the same application with STM32CubeIDE **1.16.0** this time. Everything works fine.

With regards,

rasil1127, 2024-12-19:
@ALABSTM  Hi, 
I reinstalled STM32CubeIDE and made a new project it seems as if the problem has been resolved. i appreciate your help.

ALABSTM, 2024-12-20:
ST Internal Reference: 199151

ALABSTM, 2025-02-13:
Dear all,

Since the issue has been **fixed**, as some of you have reported, please allow me to close this thread.

As explained in a previous [comment](https://github.com/STMicroelectronics/STM32CubeF4/issues/113#issuecomment-2541009295), there is **no need** to explicitly initialize the `VTOR` register, as long as the default location of the vector table on top of the flash memory is **not** intended to be changed.

Thank you all for your reports and feedback.

With regards,

closed 2025-02-13 by ALABSTM

Abhiram001-coder, 2025-04-07:
I tried the same approach of "Edit this section in system_stm32f4xx.c by doing the following 2 things. Full updated code snippet below
  1) uncommenting the "#define USER_VECT_TAB_ADDRESS" 
  2) adding the line #if defined(USER_VECT_TAB_ADDRESS)	
" @beachmiles for the application of External Loader file for the touchGFX application when i tick the both the check boxes of the "enable" and "initialize" inside the debug configuration my HAL_Delay()  is not working getting stuck inside and not working uwTick(). Kindly help me out!!

beachmiles, 2025-05-21:
> Dear all,
>
> Since the issue has been **fixed**, as some of you have reported, please allow me to close this thread.
>
> As explained in a previous [...]

I wasn't doing anything fancy at all, just created a basic boiler plate non RTOS stm32cubemx project and tried to do a simple HAL_Delay(1) in the main loop which was causing an infinites loop. I wasn't fiddling with any registers or custom bootloader's.

It sounds like there was a stm32cube fix somewhere in the past couple months that resolved this?
For anyone else still having this issue it looks like you may need to do a full uninstall of stm32cube rand then install the latest version.