STMicroelectronics/STM32CubeF4 #114: STM32CUBEMX, STM32cubeIDE and STM32cubeF4

vahidajalluian, 2022-03-15:
**Describe the set-up**
 * Custom board with STM32F429IGT6
 * STM32CUBEIDE 1.7 1.8 1.9

**Describe the bug**
I don't where the problem occurs, but in the case of F4, it happens. It fails in the system timer initialization and resets the chip.
Furthermore, there is a problem with LWIP in the new ide 1.9.

**How To Reproduce**
In the following links, the problem has been discussed.
https://community.st.com/s/question/0D53W00001PugSgSAJ/issue-with-ide-190-and-1100-with-stmh7-rtos-lwip
https://community.st.com/s/question/0D53W000011TluASAS/problem-with-cubemx-63-and-cube-f4-v262

VaZso, 2022-03-21:
It also happens using latest version of STM32CubeMX using a custom board with STM32F407ZET6.

There is a new initialization structure of Ethernet interface of firmware (V1.27.X, generated in main.c) which lacks definitions and also Eth-related type definitions have missing members.
It seems like old and new version-related code of ethernet interface would be messed up...

Furthermore, generation using older firmware (V1.26.x) also has compiling problems because of updated contents of lwip.c and ethernetif.c files...

RKOUSTM, 2022-04-08:
Hi @vahidajalluian and @VaZso,

Thank you for your contribution. You are absolutely right about this point. Actually, the point you raised has already been dealt and fixed internally. The fix will be made available in the frame of a future release of CubeMX.

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to our ecosystem (namely the Cube MX tool), please allow me to close it.

Thank you for your contribution. We are looking forward to reading from you again.

With regards,

RKOUSTM, 2022-04-08:
ST Internal Reference: 121854

closed 2022-04-08 by RKOUSTM