STMicroelectronics/STM32CubeF4 #115: Implicit downcast from uint16_t to uint8_t

daniel-starke, 2022-03-17:
**Describe the set-up**
 * Enabled `-Wconversion` in gcc.

**Describe the bug**
The compiler warns about an implicit downcast from `uint16_t` to `uint8_t` in `LL_ADC_REG_ReadConversionData8()` and `LL_ADC_REG_ReadConversionData6()`.

https://github.com/STMicroelectronics/STM32CubeF4/blob/3d6be4bd406f275728e0a321cc371c62a3100533/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h#L4016-L4019

https://github.com/STMicroelectronics/STM32CubeF4/blob/3d6be4bd406f275728e0a321cc371c62a3100533/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h#L4031-L4034

**How To Reproduce**
Compile a file which includes `stm32f4xx_ll_adc.h` with `-Wconversion` in gcc.

**Additional context**
A cast is already included for the return value within the named functions.
I propose to change the cast from  `uint16_t` to `uint8_t`.

ASELSTM, 2022-04-04:
Hi @daniel-starke,

First, thank you for this report.  Actually, the point you have mentioned has been raised within another [issue](https://github.com/STMicroelectronics/stm32f4xx_hal_driver/issues/8). It has been fixed internally and the fix will be available in the frame of a next release. 

The fix is as follow:

```diff
__STATIC_INLINE uint8_t LL_ADC_REG_ReadConversionData8(ADC_TypeDef *ADCx)
{
-  return (uint16_t)(READ_BIT(ADCx->DR, ADC_DR_DATA));
+  return (uint8_t)(READ_BIT(ADCx->DR, ADC_DR_DATA));
}
```

```diff
 __STATIC_INLINE uint8_t LL_ADC_REG_ReadConversionData6(ADC_TypeDef *ADCx) 
 { 
-  return (uint16_t)(READ_BIT(ADCx->DR, ADC_DR_DATA)); 
+  return (uint8_t)(READ_BIT(ADCx->DR, ADC_DR_DATA)); 
 } 
```

With regards,

ALABSTM, 2023-07-26:
Fixed in commit d5af56388ff037735ac99de39abf2b46f9921aa3

closed 2023-07-26 by ALABSTM