STMicroelectronics/STM32CubeF4 #116: Unused parameter warning in LL_ADC_DMA_GetRegAddr()

daniel-starke, 2022-03-17:
**Describe the set-up**
 * Enabled `-Wunused` in gcc.

**Describe the bug**
The compiler warns that the parameter `Register` in `LL_ADC_DMA_GetRegAddr()` is unsed.

https://github.com/STMicroelectronics/STM32CubeF4/blob/3d6be4bd406f275728e0a321cc371c62a3100533/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h#L1880-L1884

**How To Reproduce**
Compile a file which includes stm32f4xx_ll_adc.h with `-Wunused` in gcc. The target shall not define `ADC_MULTIMODE_SUPPORT`.

**Additional context**
I propose to add `UNUSED(Register);` at the beginning of `LL_ADC_DMA_GetRegAddr()`.

ASELSTM, 2022-03-29:
Hi @daniel-starke,

Thank you for this report. The point you raised out has been fixed within other STM32cube packages and still not yet fixed on STM32CubeF4, it will be made available in the frame of a future release.  

The fix is as follow : 

```diff
 __STATIC_INLINE uint32_t LL_ADC_DMA_GetRegAddr(ADC_TypeDef *ADCx, uint32_t Register) 
 { 
+  /* Prevent unused argument compilation warning */
+  (void)Register;

   /* Retrieve address of register DR */ 
   return (uint32_t)&(ADCx->DR); 
 } 
```

With regards,

ASELSTM, 2022-03-29:
ST Internal Reference: 125353

smartel99, 2022-05-04:
The exact same issue is also appearing in `LL_ADC_IsActiveFlag_MST_EOCS`.

https://github.com/STMicroelectronics/STM32CubeF4/blob/3d6be4bd406f275728e0a321cc371c62a3100533/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h#L4395-L4398

Would it be possible to fix it in that release as well?

Also, doesn't the HAL usually use the `UNUSED` macro to mark unused variables?

ALABSTM, 2023-07-26:
Fixed in commit d5af56388ff037735ac99de39abf2b46f9921aa3

closed 2023-07-26 by ALABSTM