STMicroelectronics/STM32CubeF4 #120: ethenet interrupt not enabled

DedkovArtem, 2022-04-01:
in the **void ethernet_link_thread(void* argument)** procedure, the ethernet is started without interrupt enabled
`if(linkchanged)
    {
...
      **HAL_ETH_Start(&heth);**
  ....
    }`
should probably be **HAL_ETH_Start_IT(&heth);** similar to how it happens in the procedure **static void low_level_init(struct netif *netif)**

ASELSTM, 2022-04-11:
Hi @DedkovArtem,

The issue you are pointing out has been fixed in the frame of the latest release v1.27.0 of the STM32CubeF4. Please allow me then to close this thread.

With regards,

DedkovArtem, 2022-04-11:
I am using Cube version 6.5.0 and STM32CubeF4 version 1.27.0. After generating the code, I get the **ethernetif.c** file with the following content:
```
void ethernet_link_thread(void* argument)
{
  ETH_MACConfigTypeDef MACConf = {0};
  int32_t PHYLinkState = 0;
  uint32_t linkchanged = 0U, speed = 0U, duplex = 0U;

  struct netif *netif = (struct netif *) argument;
/* USER CODE BEGIN ETH link init */

/* USER CODE END ETH link init */

  for(;;)
  {
  PHYLinkState = LAN8742_GetLinkState(&LAN8742);

  if(netif_is_link_up(netif) && (PHYLinkState <= LAN8742_STATUS_LINK_DOWN))
  {
    HAL_ETH_Stop_IT(&heth);
    netif_set_down(netif);
    netif_set_link_down(netif);
  }
  else if(!netif_is_link_up(netif) && (PHYLinkState > LAN8742_STATUS_LINK_DOWN))
  {
    switch (PHYLinkState)
    {
    case LAN8742_STATUS_100MBITS_FULLDUPLEX:
      duplex = ETH_FULLDUPLEX_MODE;
      speed = ETH_SPEED_100M;
      linkchanged = 1;
      break;
    case LAN8742_STATUS_100MBITS_HALFDUPLEX:
      duplex = ETH_HALFDUPLEX_MODE;
      speed = ETH_SPEED_100M;
      linkchanged = 1;
      break;
    case LAN8742_STATUS_10MBITS_FULLDUPLEX:
      duplex = ETH_FULLDUPLEX_MODE;
      speed = ETH_SPEED_10M;
      linkchanged = 1;
      break;
    case LAN8742_STATUS_10MBITS_HALFDUPLEX:
      duplex = ETH_HALFDUPLEX_MODE;
      speed = ETH_SPEED_10M;
      linkchanged = 1;
      break;
    default:
      break;
    }

    if(linkchanged)
    {
      /* Get MAC Config MAC */
      HAL_ETH_GetMACConfig(&heth, &MACConf);
      MACConf.DuplexMode = duplex;
      MACConf.Speed = speed;
      HAL_ETH_SetMACConfig(&heth, &MACConf);
      HAL_ETH_Start(&heth);
      netif_set_up(netif);
      netif_set_link_up(netif);
    }
  }

/* USER CODE BEGIN ETH link Thread core code for User BSP */

/* USER CODE END ETH link Thread core code for User BSP */

    osDelay(100);
  }
}
```

In the __void ethernet_link_thread(void* argument)__ task handler (and not in the __static void low_level_init(struct netif *netif)__  where there is similar code),which periodically monitors the presence of a connection, the enable ethernet occurs without enable interrupt and, accordingly, nothing works. What am I doing wrong?
Attached is the project file.

[testF429Eth.zip](https://github.com/STMicroelectronics/STM32CubeF4/files/8464463/testF429Eth.zip)