STMicroelectronics/STM32CubeF4 #121: [ETH] two constants specifying the size of the Rx Buff

DedkovArtem, 2022-04-01:
I use STM32F429IITx + FreeRTOS + ETH+ LWIP.
In my opinion, two constants set the same thing.

**First constant**
In the file lwipopts.h
`#define ETH_RX_BUFFER_SIZE 1536`
Its value can be configured via Cube using the Rx Buffers Length input field.

**Second constant**
In the file stm32f4xx_hal_conf.h
`#define ETH_RX_BUF_SIZE ETH_MAX_PACKET_SIZE`
which refers to the file stm32f4xx_hal_eth.h in which the constant is declared
`#define ETH_MAX_PACKET_SIZE ((uint32_t)1528U)`
its value cannot be changed via Cube

A)
ETH_RX_BUFFER_SIZE is used when describing the type RxBuff_t
ETH_RX_BUF_SIZE is used when configuring the DMARxDscrTab in the
static void procedure ETH_DMARxDescListInit(ETH_HandleTypeDef *heth)
`dmarxdesc->DESC1 = ETH_DMARXDESC_RCH | ETH_RX_BUF_SIZE;`
Both constants define the size of the same array. ETH_RX_BUFFER_SIZE allocates space in RAM for it, and ETH_RX_BUF_SIZE tells Eth controller the available buffer size.

B)
The ETH_RX_BUFFER_SIZE value (not a reference) is used in the static void low_level_init(struct netif *netif) function
`heth.Init.RxBuffLen = 1536;`
it would be logical to change to
`heth.Init.RxBuffLen = ETH_RX_BUFFER_SIZE ;`

ASELSTM, 2022-05-11:
ST Internal Reference: 128005

ASELSTM, 2023-04-06:
Hi @DedkovArtem,

Thank you for your contribution. This issue has been fixed in the frame of version v1.27.1 of the STM32CubeF4. Please allow me then to close this thread.

With regards,

closed 2023-04-06 by ASELSTM