STMicroelectronics/STM32CubeF4 #122: Broken link to STemWin middleware download

bmcdonnell-fb, 2022-04-07:
**Describe the bug**
Link to [STemWin download page](https://www.st.com/en/licensed-software/stemwin-mw.html?softwareProduct=STemWIN-MW&softwareVersion=5.44.40) in the [STemWin readme.txt](https://github.com/STMicroelectronics/STM32CubeF4/blob/b5abca20c9676b04f8d2885a668a9b653ee65705/Middlewares/ST/STemWin/readme.txt#L16) is broken (404 error page).

**Additional context**
The same is true of all `STMicroelectronics/STM32CubeF[012347]` repos here on GitHub.

If there is a new location to download the files, please provide it here in comments ASAP for our reference, in addition to updating the readme. Or tell us if you fixed the link on the website.

ALABSTM, 2022-04-08:
Hi @bmcdonnell-fb,

Thank you for having reported this point. Indeed, we are experiencing some issue with these links. We recommend to download the full zipped package from [st.com](https://www.st.com/en/embedded-software/stm32cubef4.html) as a workaround.

With regards,

bmcdonnell-fb, 2022-04-11:
The published full zipped STM32CubeF4 packages (v1.23.0 - v1.27.0) only include the latest STemWin (5.44). I need the older versions, too (5.32 & 5.40).

ALABSTM, 2022-04-13:
Hi @bmcdonnell-fb,

In this case, you can use **Cube MX** to download older versions of the STM32CubeF4 firmware package with the versions you are looking for of the STemWin library. [Here](https://github.com/STMicroelectronics/STM32CubeF4/issues/16#issuecomment-890855457)'s how to proceed.

I hope this helps. Please let me know how it went once you'd have given it a try.

With regards,

ALABSTM, 2022-04-29:
Hi @bmcdonnell-fb,

Glad to hear it helped. We are working on the subject. I will keep you informed, shall there be any progress.

With regards,

ALABSTM, 2023-03-13:
Hi @bmcdonnell-fb,

I hope you are fine. Following recent changes in our way of work, there is no longer need to use the broken link you reported. From now on, we invite users to download the full firmware package from our **st.com** website in case they want to use any library under click-thru or one of the related projects (demo, application, example).

More details are given in [this](https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/discussions/17) announcement. Please allow me to close this issue now. Thank you again for having reported.

With regards,

closed 2023-03-13 by ALABSTM

bmcdonnell-fb, 2025-05-13:
@ALABSTM please reopen this issue, update the README to align with your announcement, and fix the broken link there.