STMicroelectronics/STM32CubeF4 #123: No way to undefine LWIP_RAM_HEAP_POINTER in CubeMX

DeeFuse, 2022-04-08:
When creating a Project using CubeMX 6.5 and STM32Cube_FW_F4_V1.27.0 the LWIP driver will be broken without external RAM.
There is no way to undefine LWIP_RAM_HEAP_POINTER to allocate the LWIP-Heap in internal RAM without extensive linkerfile modifications.

The comment in `LWIP/Target/lwipopts.h` also mentions the H7 MCU for which the defualt memory address would be valid but not for an F4.
```
/*----- Default Value for H7 devices: 0x30044000 -----*/
#define LWIP_RAM_HEAP_POINTER 0x30044000 
```

DedkovArtem, 2022-04-08:
In the lwipopts.h file, in the user section, do 
`#undef LWIP_RAM_HEAP_POINTER`

DeeFuse, 2022-04-09:
Thanks for pointing the user section out.
But I think it should be possible to disable it through CubeMX and even disable it by default.
I've spent some time tracing down a HardFault condition after creating a clean test application.

ASELSTM, 2022-06-24:
Hi @DedkovArtem,

The point you reported is actually related to the **CubeMX** generation problem and not to the firmware published in this repository. Unfortunately, we don't treat aspect related to CubeMX tool in our GitHub repositories.  They are rather treated at the [STM32CubeMX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated page of the **ST Community** . 

However, the issue you pointed out has been spotted and fixed internally. The fix will be available in the frame of the next release **V6.6.0** of the **CubeMx**.

Since this issue is not directly related to the STM32Cube firmware but rather to our ecosystem, please allow me then to close this thread. Thank you for your comprehension. 

With regards,