STMicroelectronics/STM32CubeF4 #124: STM32F4 SAI using DMA will not trigger an interrupt

fyyxxm, 2022-04-10:
MCU：stm32f427
STM32CUBEMX : 6.5
HAL lib:1.27

Issue:
Using the SAI project generated by stm32cubemx, DMA will not trigger interrupt
[image]
[image]

In the initialization process, Sai is initialized first, and then DMA is initialized, which can trigger an interrupt.
Before change：

```
int main(void)
{
  /* USER CODE BEGIN 1 */

  /* USER CODE END 1 */

  /* MCU Configuration--------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* USER CODE BEGIN Init */

  /* USER CODE END Init */

  /* Configure the system clock */
  SystemClock_Config();

  /* USER CODE BEGIN SysInit */

  /* USER CODE END SysInit */

  /* Initialize all configured peripherals */
  MX_GPIO_Init();
  MX_DMA_Init();
  MX_SAI1_Init();
  /* USER CODE BEGIN 2 */

  /* USER CODE END 2 */
  HAL_SAI_Transmit_DMA(&hsai_BlockA1, test_buff, 1024);
  /* Infinite loop */
  /* USER CODE BEGIN WHILE */
  while (1)
  {
    /* USER CODE END WHILE */

    /* USER CODE BEGIN 3 */
    HAL_Delay(500);
    /* USER CODE BEGIN 3 */
  }
  /* USER CODE END 3 */
}
```
After change：
```
int main(void)
{
  /* USER CODE BEGIN 1 */

  /* USER CODE END 1 */

  /* MCU Configuration--------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* USER CODE BEGIN Init */

  /* USER CODE END Init */

  /* Configure the system clock */
  SystemClock_Config();

  /* USER CODE BEGIN SysInit */

  /* USER CODE END SysInit */

  /* Initialize all configured peripherals */
  MX_GPIO_Init();

  MX_SAI1_Init();
  MX_DMA_Init();
  /* USER CODE BEGIN 2 */

  /* USER CODE END 2 */
  HAL_SAI_Transmit_DMA(&hsai_BlockA1, test_buff, 1024);
  /* Infinite loop */
  /* USER CODE BEGIN WHILE */
  while (1)
  {
    /* USER CODE END WHILE */

    /* USER CODE BEGIN 3 */
    HAL_Delay(500);
    /* USER CODE BEGIN 3 */
  }
  /* USER CODE END 3 */
}
```

But it can only be triggered once. My DMA transmission mode is configured with ：
`hdma_ sai1_ a.Init. Mode = DMA_ CIRCULAR; `
So there's still a problem.

ASELSTM, 2022-06-24:
Hi @fyyxxm,

Would you please share the whole project you have used to reproduce the issue in order to allow a better analysis of the problem.

With regards,

ASELSTM, 2022-09-30:
Hi @fyyxxm,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2022-09-30 by ASELSTM