STMicroelectronics/STM32CubeF4 #135: Prefetch buffer check logic only works for rev 0x1001

atdicapf, 2022-06-07:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
STM32F324XG eval based board with a "Z" rev micro.
KEIL 5.29
Example LWIP and Freertos Server Socket Project

**Describe the bug**
On a revision 0x1007 micro the prefetch cannot be enabled due to the logic in the SystemClock_Config disabling the prefetch.
It is only checking for revision == 0x1001.
According to the eratta any version 0x1001 and above should be able to use the prefetch.

Specific lines:

`  /* STM32F405x/407x/415x/417x Revision Z and above devices: prefetch is supported  */  
  if (HAL_GetREVID() == 0x1001)  
  {  
    /* Enable the Flash prefetch */  
    __HAL_FLASH_PREFETCH_BUFFER_ENABLE();  
  }  `

**How To Reproduce**
1. Load Up Project
2. Read code at: SystemClock_Config
3. Check generated code

**Additional context**
I believe the logic should be as:
`  if (HAL_GetREVID() >= 0x1001)
  {
    /* Enable the Flash prefetch */
    __HAL_FLASH_PREFETCH_BUFFER_ENABLE();
  }`

**Screenshots**
See eratta:
[image]
[image]

ASELSTM, 2022-06-17:
ST Internal Reference: 130480

TOUNSTM, 2023-11-23:
Hello @atdicapf,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.28.0 of the STM32CubeF4 published recently on GitHub. Thank you again for having reported.

With regards,

closed 2023-11-23 by TOUNSTM