STMicroelectronics/STM32CubeF4 #138: STM32412G-Discovery BSP : Undefined behavior when `BSP_COM_Init` is called with `COM2`

supcik, 2022-07-26:
**Setup**

- Board : STM32F412G-DISCO
- IDE : VS Code + Platform.io

**Describe the bug**

The BSP for the STM32F412G-DISCO (`Drivers/BSP/STM32412G-Discovery/stm32412g_discovery.c`) has a [method to initialize the serial communication](https://github.com/STMicroelectronics/STM32CubeF4/blob/52757b5e33259a088509a777a9e3a5b971194c7d/Drivers/BSP/STM32412G-Discovery/stm32412g_discovery.c#L463) :

```c
void BSP_COM_Init(COM_TypeDef COM, UART_HandleTypeDef *huart)
```

According to the documentation, `COM` can be `COM1` or `COM2`:

```c
  * @param  COM: COM port to be configured.
  *          This parameter can be one of the following values:
  *          COM1 
  *          COM2 
  * @param  huart: Pointer
```

But actually, only `COM1` ([which is used for USART2](https://github.com/STMicroelectronics/STM32CubeF4/blob/52757b5e33259a088509a777a9e3a5b971194c7d/Drivers/BSP/STM32412G-Discovery/stm32412g_discovery.h#L260)) can be used. Indeed, the number of `COM` [is set to 1](https://github.com/STMicroelectronics/STM32CubeF4/blob/52757b5e33259a088509a777a9e3a5b971194c7d/Drivers/BSP/STM32412G-Discovery/stm32412g_discovery.h#L257):

```c
#define COMn ((uint8_t)1)
```

and the arrays `COM_USART[COMn]`, `COM_TX_PORT[COMn]`, `COM_RX_PORT[COMn]`, `COM_TX_PIN[COMn]`, `COM_RX_PIN[COMn]`, `COM_TX_AF[COMn]`, `COM_RX_AF[COMn]` are only defined for a single entry : `COM1`.

So when `BSP_COM_Init` is called with `COM2` for the `COM` argument, the result is undefined.

**How To Reproduce**

1. Call `BSP_COM_Init` with `COM2` as first argument

**Additional context**

- `BSP_COM_Init` currently returns `void`, so it can't return an error if called with `COM2`. Perhaps it would be good to return a `HAL_StatusTypeDef` instead of `void`. This could also be used to detect other problems during the initialization of the USART.
- It was not intuitive to me that `COM1` was actually used for `USART2`.

**Screenshots**

n/a

ASELSTM, 2022-08-08:
ST Internal Reference: 132859

TOUNSTM, 2023-11-23:
Hello @supcik,

The issue you reported has been fixed in the frame of version **v1.28.0** of the **STM32CubeF4** published recently on GitHub. Thank you again for having reported.

With regards,

closed 2023-11-23 by TOUNSTM