STMicroelectronics/STM32CubeF4 #140: STM32CUBEMX FreeRTOS "USE_TICKLESS_IDLE -> Built in functionality enabled" generates weird code

fishjimi, 2022-08-26:
**Describe the set-up**
 * The board: custom board with STM32F446RET6
 * STM32CUBEMX 6.6.1 with STM32Cube FW_F4 V1.27.1.

**Describe the bug**
When using "USE_TICKLESS_IDLE -> Built in functionality enabled",
MCU is not placed into the low power state

**How To Reproduce**
1. turn on "USE_TICKLESS_IDLE" and chose "Built in functionality enabled"
2. regenerate code.

**Additional context**
There generated code makes ulExpectedIdleTime = 0,  which makes MCU never enter low power mode.
Why the macro being defined in this way?
```C
//FreeRTOSConfig.h
#define configPRE_SLEEP_PROCESSING(__x__)                           \
                                      do{                           \
                                         __x__ = 0;                 \
                                         PreSleepProcessing(__x__); \
                                      }while(0)
```
allso see 
[https://community.st.com/s/question/0D53W00000beBVnSAM/latest-cubemx-update-breaks-freertos-tickless-idle](https://community.st.com/s/question/0D53W00000beBVnSAM/latest-cubemx-update-breaks-freertos-tickless-idle)

ASELSTM, 2022-09-14:
Hi @fishjimi,

I hope you are fine. The issue you reported has been logged into an internal bug tracker it will be managed by the development team.

However, as it is related to the ecosystem (namely the CubeMX tool) rather than the firmware published within this repository (CMSIS, HAL, BSP, etc.), please allow me thus to keep this thread closed. 

In case you have other CubeMX-related issues to share, please submit them in the [ST Community](https://community.st.com/s/question/0D53W00000JhsU4SAJ/stm32cubemx-ide-issue-throws-warning-when-using-macros) in the CubeMX related section. There you shall find people who will address your request and whom you will hopefully find a solution with.

Thank you for your comprehension and thank you again for having reported.

With regards,

ASELSTM, 2022-09-14:
ST Internal Reference: 134667

closed 2022-09-14 by ASELSTM

XXX-01, 2023-11-08:
Excuse me, is there an answer to this question? I had this problem too.

fishjimi, 2023-11-08:
> Excuse me, is there an answer to this question? I had this problem too.

@XXX-01 You can modify this macro code yourself, if you don't know how to do, just create a F103 project and copy

XXX-01, 2023-11-08:
@fishjimi 
OK, thank you very much. 
I also found that the F103 code could be implemented, but what I didn't understand was that the F4 version of the code had to be like this.