STMicroelectronics/STM32CubeF4 #147: Check value of pllm in function HAL_RCC_GetSysClockFreq of stm32f4xx_hal_rcc.c:911

PwnVerse, 2022-10-29:
# Check value of pllm in function HAL_RCC_GetSysClockFreq of stm32f4xx_hal_rcc.c:911

Hi Community,

There seems to be a bug in the latest version of STM32CubeF4, [stm32f4xx_hal_rcc.c line 911](https://github.com/STMicroelectronics/STM32CubeF4/blob/HEAD/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c#L911). A division by zero is possible given RCC->PLLCFGR is 0. 

```c
__weak uint32_t HAL_RCC_GetSysClockFreq(void)
{
  uint32_t pllm = 0U, pllvco = 0U, pllp = 0U;
  uint32_t sysclockfreq = 0U;

  /* Get SYSCLK source -------------------------------------------------------*/
  switch (RCC->CFGR & RCC_CFGR_SWS)
  {
    case RCC_CFGR_SWS_PLL:  /* PLL used as system clock  source */
    {
      /* PLL_VCO = (HSE_VALUE or HSI_VALUE / PLLM) * PLLN
      SYSCLK = PLL_VCO / PLLP */
      pllm = RCC->PLLCFGR & RCC_PLLCFGR_PLLM;
      if(__HAL_RCC_GET_PLL_OSCSOURCE() != RCC_PLLSOURCE_HSI)
      {
        /* HSE used as PLL clock source */
        pllvco = (uint32_t) ((((uint64_t) HSE_VALUE * ((uint64_t) ((RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos)))) / (uint64_t)pllm);
      }
      else
      {
        /* HSI used as PLL clock source */
        pllvco = (uint32_t) ((((uint64_t) HSI_VALUE * ((uint64_t) ((RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos)))) / (uint64_t)pllm);
      }
      pllp = ((((RCC->PLLCFGR & RCC_PLLCFGR_PLLP) >> RCC_PLLCFGR_PLLP_Pos) + 1U) *2U);

      sysclockfreq = pllvco/pllp;
      break;
    }
  return sysclockfreq;
}
```

### Triggering the Bug

The bug can be triggered if switch case `RCC_CFGR_SWS_PLL` is reached and `RCC->PLLCFGR` is set to 0. 

### Possible Fixes

Handling the value of pllm by checking it against 0 can be one of the possible fixes.

If you feel that this bug is valid, I'd be happy to send over a PR as well :).

RJMSTM, 2023-03-01:
ST Internal Reference: 146728

RJMSTM, 2023-04-14:
Hello @PwnVerse 

This issue is duplicated of [#154](https://github.com/STMicroelectronics/STM32CubeF4/pull/154)

closed 2023-04-14 by RJMSTM