STMicroelectronics/STM32CubeF4 #151: PWR_CSR_EWUP3_Msk in stm32f412vx.h is (1 << PWR_CSR_EWUP1_Pos)

lkolbly, 2023-01-03:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
In the STM32F412Vx CMSIS header file.

**Describe the bug**

PWR_CSR_EWUP3_Msk in stm32f412vx.h is `(1 << PWR_CSR_EWUP1_Pos)`, when I think it should be `(1 << PWR_CSR_EWUP3_Pos)` (note the 1 changing to a 3). `PWR_CSR_EWUP3_Pos` is defined but is not used anywhere else

Link to the file (`Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f412vx.h`): https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f412vx.h around line 8950 (at least, on the commit I'm using, which is slightly older.)

**How To Reproduce**

Look at the source code.

**Additional context**
I think it's just a typo somewhere, not sure if this code gets auto-generated or anything.

**Screenshots**
None

HBOSTM, 2023-02-02:
Hello @lkolbly ,

This point has been reported to our development teams for deeper analysis. We will get back to you as soon as we have updates.

With Regards,

RJMSTM, 2023-04-14:
Hello @lkolbly 

Thank you again for having reported this point. It has been fixed in the frame of version 1.28.0 of this firmware.

With regards,

closed 2023-11-23 by TOUNSTM