STMicroelectronics/STM32CubeF4 #152: STM32F4 STBITERR incorrectly handled

keck9939, 2023-02-08:
In the file `stm32f4xx_hal_sd.c`, in the routine `HAL_SD_ReadBlocks`, if the flag `SDIO_FLAG_STBITERR` is set in the controller, the transfer will stop (assuming it ever really started), but the error is not correctly returned to the caller. There needs to be another `else if` down below the comment `/* Get error state */` to handle this case. As it is, if a `STBITERR` occurs, the return value will be `HAL_OK` and the caller will think the routine completed successfully when in fact it failed.
The same error also occurs in `HAL_SD_WriteBlocks`.

HBOSTM, 2023-03-02:
ST Internal Reference: 146826

HBOSTM, 2023-03-02:
Hello @keck9939  ,

Thank you for your contribution.
This point has been reported to our development teams. I will get back to you as soon as I have any updates.

With Regards,

TOUNSTM, 2023-11-23:
Hello @keck9939,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.28.0 of the STM32CubeF4 published recently on GitHub. Thank you again for having reported.

With regards,

closed 2023-11-23 by TOUNSTM