STMicroelectronics/STM32CubeF4 #153: GPIO toggling led not working after use Bypass clock source

ahsanu123, 2023-02-12:
hey all, 
recently i started new project (STM32F407VGT6) use USB FS with stm32 CUBE IDE, i choose clock source from bypass(cause usb cant use internal clock), after code is generated by cube ide, i check by blink led on PA1, but the led is not blinking at all. but after i disable USB and change clock source to internal, led is blinking as it should.

am i wrong with configuration of USB and clock source?, (i already watch usb tutorial on stm32 youtube channel also)

i believe my code is not wrong (cause its just basic blink led):

**here is my board** https://github.com/mcauser/MCUDEV_DEVEBOX_F407VGT6
```c++
int main(void)
{
  /* USER CODE BEGIN 1 */

  /* USER CODE END 1 */

  /* MCU Configuration--------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* USER CODE BEGIN Init */

  /* USER CODE END Init */

  /* Configure the system clock */
  SystemClock_Config();

  /* USER CODE BEGIN SysInit */

  /* USER CODE END SysInit */

  /* Initialize all configured peripherals */
  MX_GPIO_Init();
  /* USER CODE BEGIN 2 */

  HAL_GPIO_WritePin(led_GPIO_Port, led_Pin, GPIO_PIN_RESET);

  /* USER CODE END 2 */

  /* Infinite loop */
  /* USER CODE BEGIN WHILE */
  while (1)
  {
	  HAL_GPIO_TogglePin(led_GPIO_Port, led_Pin);
	  HAL_Delay(100);
    /* USER CODE END WHILE */

    /* USER CODE BEGIN 3 */
  }
  /* USER CODE END 3 */
}

```
any help is really appreciated, thanks

markdlehane, 2023-02-12:
Hi.  I can't see how you're initialising led_GPIO_port/led_Pin.  Is this done in the MX_GPIO_Init() call and if so, what is the initialisation code?

You should have some code such as the following example: 

__GPIOA_CLK_ENABLE(), 
GPIO_InitStruct.Pin = led_Pin;   
GPIO_InitStruct.Mode = GPIO_MODE_EVT_RISING;   
HAL_GPIO_Init(led_GPIO_port, &GPIO_InitStruct);

I would check for this first.

Hope it helps.

ahsanu123, 2023-02-12:
@markdlehane thanks for your reply, i already intialized gpio for led, here is code.(i also change Mode to `GPIO_MODE_EVT_RISING` but nothing change).

i also try to make new project with only usb and gpio enabled, its working, but after i change to my other project with some peripheral enabled, usb and toggling gpio is not working, is peripheral usage with usb have limitation (i need disabled some peripheral maybe?)

i also enabled serial wire for debuging (does this have any effect?)

```c++
static void MX_GPIO_Init(void)
{
  GPIO_InitTypeDef GPIO_InitStruct = {0};

  /* GPIO Ports Clock Enable */
  __HAL_RCC_GPIOH_CLK_ENABLE();
  __HAL_RCC_GPIOC_CLK_ENABLE();
  __HAL_RCC_GPIOA_CLK_ENABLE();
  __HAL_RCC_GPIOB_CLK_ENABLE();
  __HAL_RCC_GPIOD_CLK_ENABLE();

  /*Configure GPIO pin Output Level */
  HAL_GPIO_WritePin(led_GPIO_Port, led_Pin, GPIO_PIN_RESET);

  /*Configure GPIO pin : led_Pin */
  GPIO_InitStruct.Pin = led_Pin;
  GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
  GPIO_InitStruct.Pull = GPIO_NOPULL;
  GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_LOW;
  HAL_GPIO_Init(led_GPIO_Port, &GPIO_InitStruct);

}
```

iam curious if initialization sequence is have effect, cause after i change the init sequence my computer can recognize the usb i plugged in (at least its can detect it) but still go error like screenshot below

[image]

```c++
/* Initialize all configured peripherals */
  MX_GPIO_Init();
  MX_USB_DEVICE_Init();  //<===== this function
  MX_ADC1_Init();
  MX_I2C1_Init();
  MX_SDIO_SD_Init();
  MX_SPI1_Init();
  MX_USART3_UART_Init();
  MX_TIM2_Init();
                          // <======= first is over here
  /* USER CODE BEGIN 2 */
```

fpistm, 2023-02-13:
Hi,
this "issue" was also reported on stm32duino forum:
https://www.stm32duino.com/viewtopic.php?p=11683#p11683
Original issue that bypass does not working is simply that the mentioned board uses a HSE xtal.

HBOSTM, 2023-02-17:
Hello @ahsanu123,

Firstly, thank you for this contribution. 
On my side there is no problem in this example. So, would you please give me more details about how you got this issue? And share all of the code you have used to reproduce this issue.

With regards,

ahsanu123, 2023-02-18:
thanks for replying, 
after a day of trying change setting on code, my problem was on SDIO peripheral, after i disable initialization of SDIO, code is working as expected. i don't undestand why its happen.

but this my full code made with cube ide: 
https://github.com/ahsanu123/usb_cdc

cube ide version 1.11.2
[image]

closed 2023-12-15 by ALABSTM