STMicroelectronics/STM32CubeF4 #158: __HAL_RCC_APB2_FORCE_RESET cause mcu reset!

Krislanqx, 2023-07-03:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * STM32F407ZGT6 with external flash
 *KEIL MDK 5.26

**Describe the bug**
#define __HAL_RCC_APB2_FORCE_RESET()     (RCC->APB2RSTR = 0xFFFFFFFFU)

**How To Reproduce**
Indicate the global behavior of your application project
A small FLASH programmer is running in internal RAM (DTCMRAM @ 0x20000000). After it programs the external flash it unloads itself and calls HAL_DeInit() which calls __HAL_RCC_AHB3_FORCE_RESET(). At this point the microprocessor resets.

The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)
The __HAL_RCC_APB2_FORCE_RESET() resides in stm32h7xx_hal_rcc.h

The use case that generates the problem
Calling the HAL_DeInit() when the program prepares to unload itself.

How we can reproduce the problem
I don't know all conditions which makes the function fail. I suspect it has something to do with calling the function from internal RAM. I don't think you need to reproduce the problem. It is clearly a bug (see below) which should be rectified.

**Additional context**
The problem may be the same as the link. Please refer to it.
https://github.com/STMicroelectronics/STM32CubeH7/issues/19

**Screenshots**
None

RJMSTM, 2023-07-04:
Hello @Krislanqx,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

Krislanqx, 2023-07-15:
@RJMSTM @TOUNSTM 
According to the RM0090 reference manual，Bits 31:8  must be kept at reset value and should not be set. But in the STM32CubeF4, RCC_AHB2RSTR = 0xFFFFFFFFU.
[image]

HBOSTM, 2023-08-03:
ST Internal Reference: 158724

Krislanqx, 2023-10-25:
@HBOSTM Hi，HBOSTM: I would like to know when this issue can be closed and new version release.

TOUNSTM, 2023-11-23:
Hello @Krislanqx,

The issue you reported has been fixed in the frame of version **v1.28.0** of the **STM32CubeF4** published recently on GitHub. Thank you again for having reported.

With regards,

closed 2023-11-23 by TOUNSTM