STMicroelectronics/STM32CubeF4 #160: STM32F429 + LAN8742a (custom board) problem with lwip

dany137718, 2023-07-15:
hi guys!
I have a problem with my stm32. LWIP goes to software reset and returns HAL_ETH_ERROR_TIMEOUT error to heth->ErrorCode and my error handle is called after that. 
I check my REF_CLK pin on LAN9742 and there is no signal. 

my question is: what is the problem? why do I have no signal? is my hardware wrong? 

thanks.

RJMSTM, 2023-07-20:
Hello @dany137718,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

ASEHSTM, 2024-08-19:
Hello @dany137718,

Sorry for the delayed reply. Could you please specify the version of the HAL driver you are using?

With Regards,

ASEHSTM, 2025-03-24:
Hello @dany137718,

Any update regarding this thread?

With Regards,

ASEHSTM, 2025-07-28:
Hello, 

Please allow me to close this thread as no activity.

With Regards,