STMicroelectronics/STM32CubeF4 #163: Unnecessary 64-bit calculations result in 700+ bytes increase in firmware size

yakov-bakhmatov, 2023-09-04:
Function `uint32_t HAL_RCC_GetSysClockFreq(void)` calculates expression `PLL_VCO = (HSE_VALUE or HSI_VALUE / PLLM) * PLLN` using 64-bit multiplication and division.

https://github.com/STMicroelectronics/STM32CubeF4/blob/d5af56388ff037735ac99de39abf2b46f9921aa3/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c#L905-L920

This forces the compiler (in particular gcc) to link to an external `__aeabi_uldivmod` function that performs a 64-bit division.

But to calculate the expression `a * b / c`, where `a`, `b` and `c` are `uint32_t` and the result is also 32 bits, it is possible without expanding to 64 bits.

Let `a = m * c + n`, `b = p * c + q`. Then

```
a * b / c = (m * c + n) * (p * c + q) / c =
  (m * p * c * c + m * q * c + n * p * c + n * q) / c =
  m * p * c + m * q + n * p + n * q / c
```

Define the auxiliary function:

```
static uint32_t muldiv(uint32_t a, uint32_t b, uint32_t c) {
    uint32_t m = a / c;
    uint32_t n = a % c;
    uint32_t p = b / c;
    uint32_t q = b % c;
    return m * p * c + m * q + n * p + n * q / c;
}
```

Expressions in lines 911, 916 are converted to the following

```
-        pllvco = (uint32_t) ((((uint64_t) HSE_VALUE * ((uint64_t) ((RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos)))) / (uint64_t)pllm);
+        pllvco = muldiv(HSE_VALUE, (RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos, pllm);
```

```
-        pllvco = (uint32_t) ((((uint64_t) HSI_VALUE * ((uint64_t) ((RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos)))) / (uint64_t)pllm);
+        pllvco = muldiv(HSI_VALUE, (RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos, pllm);
```

How this change affects the size of the binary.

For example, create an empty Makefile project in CubeMX for MCU STM32F407 and compile it by arm-gnu-toolchain-12.2.

```
arm-none-eabi-size build/stm32f407-empty.elf

   text	   data	    bss	    dec	    hex	filename
   3668	     20	   1572	   5260	   148c	build/stm32f407-empty.elf
```

Using the `muldiv` function:

```
arm-none-eabi-size build/stm32f407-empty.elf

   text	   data	    bss	    dec	    hex	filename
   2892	     20	   1572	   4484	   1184	build/stm32f407-empty.elf
```

The difference in binary size is 776 bytes.

Alec-Davis-NZ, 2024-04-10:
ChatGPT's answer
```
#include <stdint.h>

uint32_t multiply_and_divide(uint32_t a, uint32_t b, uint32_t c) {
    // Decompose a and b
    uint32_t m = a / c;  // Integer quotient of a divided by c
    uint32_t n = a % c;  // Remainder of a divided by c
    uint32_t p = b / c;  // Integer quotient of b divided by c
    uint32_t q = b % c;  // Remainder of b divided by c

    // Perform the multiplication without overflowing
    uint32_t product = (m * c + n) * (p * c + q);

    // Perform the division
    uint32_t result = product / c;

    return result;
}

```

In this implementation, a, b, and c are all assumed to be 32-bit unsigned integers (uint32_t). We use integer division (/) and modulo (%) to calculate the values of m, n, p, and q. Then, we perform the multiplication (m * c + n) * (p * c + q) and divide by c to get the final result, ensuring that all intermediate calculations and the final result remain within the bounds of 32-bit unsigned integers.

TOUNSTM, 2024-06-21:
See Also https://github.com/STMicroelectronics/stm32l0xx_hal_driver/issues/12

bmcdonnell-fb, 2024-06-21:
> For example, create an empty Makefile project in CubeMX for MCU STM32F407 and compile it by arm-gnu-toolchain-12.2.
>
> ```
> arm-none-eabi-size build/stm32f407-empty.elf
>
>    text	   data	    bss [...]

What if you do some `uint64_t` calculation somewhere else in your application?

bmcdonnell-fb, 2024-10-09:
Code space aside, @yakov-bakhmatov's implementation is faster.

bmcdonnell-fb, 2024-10-09:
BTW, IMO it's easier to see how it works w/ more descriptive variable names, e.g.

```
static uint32_t muldiv_u32(uint32_t mul1, uint32_t mul2, uint32_t denom)
{
    uint32_t quot1 = mul1 / denom;
    uint32_t rem1  = mul1 % denom;
    uint32_t quot2 = mul2 / denom;
    uint32_t rem2  = mul2 % denom;
    return ((quot1 * quot2 * denom) + (quot1 * rem2) + (rem1 * quot2) + (rem1 * rem2 / denom));
}
```

closed 2024-11-28 by TOUNSTM

bmcdonnell-fb, 2026-10-27:
NB: the optimized function proposed here does not give correct answers for the full range of possible `uint32_t` output values. Maybe it does for all the inputs used here, though - I didn't check.