STMicroelectronics/STM32CubeF4 #166: lwIP project could not work in stm32cubeide

taehyun9203, 2023-10-16:
Hello, I had a problem with the lwIP project not working on STM32CubeIDE using the Nucleo-F429ZI board.

The cause of the problem is that the default LWIP_RAM_HEAP_POINTER default value is 0x30044000, and if STM32IDE generates code with that value, it is accessing a non-existent memory area, resulting in a hard fault.

So the solution can be summarized in two ways.
The first is to comment LWIP_RAM_HEAP_POINTER in lwipopts.h among the codes responsible for the lwip function (Figure 1).
The second is to disable ICMP checksum-related features in the Ethernet Configuration window (Figure 2).

Figure 1
[image]

Figure 2
[image]

ASEHSTM, 2024-05-13:
Hello @taehyun9203,

Thank you for your report. You are Right. However, this issue has been rectified in version 6.6.0 of STM32cubeMX.

With Regards,

ASEHSTM, 2024-05-13:
Hello @taehyun9203,

Please allow me to close this thread. You may reopen it at any time if you have any details to share with us. 
Thank you 

With regards,

closed 2024-05-13 by ASEHSTM