STMicroelectronics/STM32CubeF4 #172: BUG : Wrong check of file size in the ymodem.c file

sfeutrier, 2024-04-26:
https://github.com/STMicroelectronics/STM32CubeF4/blob/6d620b9ee55ff248d195ab6ea39ee82728ff30ad/Projects/STM324x9I_EVAL/Applications/IAP/IAP_Main/Src/ymodem.c#L358C26-L358C33

In the function "Ymodem_Receive", code checks the size of the file using `if (*p_size > (USER_FLASH_SIZE + 1))` but p_size is the pointer to the output size and at this moment it is 0.
The "if" should check the "filesize" and then do `if (*filesize > (USER_FLASH_SIZE + 1))`

TOUNSTM, 2024-04-29:
ST Internal Reference: 180274

RJMSTM, 2024-06-20:
Hello,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.28.1 of the STM32CubeF4 published recently on GitHub. Thank you again for having reported.

Regards,

closed 2024-06-20 by RJMSTM