STMicroelectronics/STM32CubeF4 #174: not nice formatted

CanastraRF, 2024-06-04:
Hi
Some files for example stm32f479xx.h are not nice formatted.
the IRQ enum looks like this

**...
..CRYP_IRQn                   = 79,     /*!< CRYP crypto global interrupt                                      */
..HASH_RNG_IRQn               = 80,     /*!< Hash and Rng global interrupt                                     */
..FPU_IRQn                    = 81,     /*!< FPU global interrupt                                              */
........................................UART7_IRQn                  = 82,     /*!< UART7 global interrupt                                            */
........................................UART8_IRQn                  = 83,     /*!< UART8 global interrupt                                            */
........................................SPI4_IRQn                   = 84,     /*!< SPI4 global Interrupt                                             */
...**

Please improve this.

ASEHSTM, 2024-06-06:
Hello @CanastraRF,

Hello, thank you for your report. This will be fixed.

With Regards,

ASEHSTM, 2024-06-10:
ST Internal Reference: 183772