STMicroelectronics/STM32CubeF4 #176: CubeMX does not handle custom PHY anymore

theadib, 2024-06-06:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.
CubeMX codegenerator > 6.4
STM32F437 and Phy KSZ8081

**Describe the bug**
A clear and concise description of what the bug is.
CubeMX does not handle custom PHY anymore.
on older CubeMX, up to version 6.4, CubeMX could handle customization of PHY registers by providing some address and pattern. That can be adapted regarding the used PHY.
Now, with CubeMX > 6.4,  a BSP is needed.
Only 2(two) PHY BSP are provided.
There is no documentation howto integrate own custom driver.
The generated code makes it very hard for a good customization.
The generated lwip.c should call using abstract function name (like phy->isLink()) rather than insert the name of that choosen driver in "dp83...._link()"

**How To Reproduce**
1. Indicate the global behavior of your application project.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).

3. The use case that generates the problem.
please provide:
- a documentation howto create a custom PHY driver that is selecteable in the CubeMX
- a selecteable "custom" driver that can be extended by the user by placng the known USER sections in the source code

5. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

I also put something in the forum, but since month there is no response from ST nor it is changed in the CubeMX
For me this is rather frustrating.

**Screenshots**
If applicable, add screenshots to help explain your problem.

theadib, 2024-06-08:
Hi ALABSTM, I saw you assigned and added labels ...
it is related to Ethernet not USB. 
please change labels of the ticket from **USB** to **Ethernet**
THanks, Adib.

ALABSTM, 2024-06-11:
Hi @theadib,

Thank you for the heads-up. Label updated. Your request has been logged internally. I will keep you informed.

With regards,

ALABSTM, 2024-06-11:
ST Internal Reference: 183949

ALABSTM, 2024-06-27:
Hi @theadib,

Our development teams acknowledged your point. An update should be available in the frame of version **6.13.0** of STM32Cube**MX**.

As this issue is not directly related to the firmware published within this repository, please allow me to close it. Thank you for your comprehension and thank you again for having reported.

With regards,

closed 2024-06-27 by ALABSTM

ALABSTM, 2025-06-16:
Hi @theadib,

I hope you are fine. Some updates regarding your request. It has not been addressed yet. However, in version **6.14.0** a similar request has been addressed for the **STM32H7** series.

* From the **LwIP** configuration pane, the drop-down list in the **Platform Settings** tab now proposes the option `USER_PHY` as shown in the first screenshot below.
* If selected, the generated project will integrate a `eth_custom_phy_interface.c` file, as shown in the second screenshot, that the user can **customize** according to their needs.

In a future version of the tool, the solution should also apply to the STM32F4 series and other series too.

With regards,

[image]

[image]

ALABSTM, 2025-08-01:
Hello @theadib,

Just to let you know that, besides the basic initialization and de-initialization functions, the `eth_custom_phy_interface.c` file will also:
* be enriched with functions to allow bus IO registration and loopback mode enablement and disablement among a a handful of other things,
* expose some registers allowing to initialize the PHY element and perform a reset, among other operations.

This will be available in a future version of STM32CubeMX. We cannot share a date for the moment. Thank you for your comprehension.

With regards,