STMicroelectronics/STM32CubeF4 #180: USB MTP driver. Multiple bugs

Kun-Kun, 2024-08-27:
Certainly a bug:
https://github.com/STMicroelectronics/STM32CubeF4/blob/82738194a5e85ee5331975b04905daa0a17053ec/Middlewares/ST/STM32_USB_Device_Library/Class/MTP/Src/usbd_mtp_storage.c#L463
At the file transfer such statement calculates buffer size in a wrong way. It always 12 bytes since the header is always less than Header+data.
I offer to remove this line and to invoke as
`  (void)USBD_LL_Transmit(pdev, MTPInEpAdd, buf, len);
`
Because at file transfer there might be a situation when there is no header and the last packet of data has length 0-12 bytes. 

The second bug is on the:
https://github.com/STMicroelectronics/STM32CubeF4/blob/82738194a5e85ee5331975b04905daa0a17053ec/Middlewares/ST/STM32_USB_Device_Library/Class/MTP/Src/usbd_mtp_storage.c#L109
It invokes the function that writes into buffer and sets it size.
But at line: 
https://github.com/STMicroelectronics/STM32CubeF4/blob/82738194a5e85ee5331975b04905daa0a17053ec/Middlewares/ST/STM32_USB_Device_Library/Class/MTP/Src/usbd_mtp_storage.c#L113
We overwrite this buffer with 12 byte header at the start of the file transfer thus corrupts 12 bytes of provided data in the buffer.
I offer to change line 109 with:
`(void)((USBD_MTP_ItfTypeDef *)pdev->pUserData[pdev->classId])->ReadData(hmtp->OperationsContainer.Param1,
                                                                              (uint8_t *)data_buff+MTP_CONT_HEADER_SIZE, &MTP_DataLength);`
  As a result move the pointer by 12 bytes and write buffer in the right section
  And it is good to notice in ReadData comments that it should return buffer with size 52 at first invocation

ALABSTM, 2024-09-24:
Hi @Kun-Kun,

Thank you for all these details and proposals. Did you experience failures caused by what you reported as bugs, or are you reporting assumptions based on code inspection?

In case you faced real failures, could you please share a code snippet or a minimal project allowing to reproduce them please?

Thank you,

Kun-Kun, 2024-10-27:
Hi @ALABSTM 
I did face issues during the implementation of the MTP on STM32F4DISCOVERY. I had to fix code problems above in this repository to make it work
https://github.com/Kun-Kun/STM32F407-MTP-MemMap
Unfortunately I have no time to make a more minimal project to prove that it doesn't work with current driver version.
But you can simply rollback [usbd_mtp_storage.c](https://github.com/STMicroelectronics/STM32CubeF4/blob/82738194a5e85ee5331975b04905daa0a17053ec/Middlewares/ST/STM32_USB_Device_Library/Class/MTP/Src/usbd_mtp_storage.c#L463) file in this project and verify that it won't transfer a data.

ALABSTM, 2024-11-06:
Hi @Kun-Kun,

I forwarded you report to our development teams. I will keep you informed.

With regards,

ALABSTM, 2025-07-28:
Fixed in release **1.28.2**

closed 2025-07-28 by ALABSTM