STMicroelectronics/STM32CubeF4 #181: STM32CUBEIDE optimization level on USB CDC mode

vahidajalluian, 2024-08-31:
Dear Developers,

I am encountering an error and would like to determine if it is related to my code or an issue with the ST USB library.

Initially, I suspected that the problem might be caused by another part of my program, which is quite large and uses FreeRTOS. However, after simplifying the program as shown below, it appears that the issue lies within the ST USB code library.

I am using an STM32F429BIT6 chip with STM32CubeMX version 6.12 and F4 pack version 1.28.1. The core clock is set to 168MHz with a 25MHz external clock, and the USB clock is set to 48MHz. I added USB functionality, trying both HS and FS modes (individually and simultaneously), and included CDC support.

The problem arises when I use any optimization level (O1 to O3) in my program. Without optimization, the same code works correctly, whether FreeRTOS is enabled or not. With optimization, however, the mentioned code fails to work, regardless of whether FreeRTOS is used. In the working scenario, the PC successfully enumerates the USB device, and data transfer is fine. In the non-working scenario, Windows detects some activity on the ports, but it doesn’t show any device or even an error message indicating a failed descriptor request—only a refresh of the Device Manager screen.

I have searched the forums and found that some variables in the library might need to be declared with the volatile keyword but are not.

Does anyone have suggestions on how to resolve this issue?
The link to the forum is 
https://community.st.com/t5/stm32-mcus-embedded-software/stm32cubeide-usb-optimization-level-on-usb-cdc-mode/td-p/714644

ALABSTM, 2024-09-23:
Hi @vahidajalluian,

Could you please share a minimal project allowing to reproduce the issue you're describing please? A project that would work if no optimization is applied and fail otherwise. Ideally, if you're using STM32CubeMX to generate this project, the IOC file would be welcome.

Thank you in advance,

ALABSTM, 2024-09-23:
Hi @vahidajalluian,

According to our development teams, the point you mentioned from the forum is rather related to the `HAL` driver than to the `MW` library. Moreover, the `__IO` or `volatile` related issue has already been fixed, as you can see, for instance, in [this](https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/064b123f8fc187d468bb762cc150d198f873478d/Src/stm32f4xx_ll_usb.c#L1406-L1419) code snippet.

Hence, this looks more like an issue related to your application. As indicated in the [CONTRIBUTING.md](https://github.com/STMicroelectronics/STM32CubeF4/blob/master/CONTRIBUTING.md#3-support-requests-and-questions) file, I suggest that you submit this to the [ST Community](https://community.st.com/t5/stm32-mcus-embedded-software/bd-p/mcu-embedded-software-forum). There, you will hopefully find the support you are looking for.

Please allow me to close this issue. Thank you for your comprehension and best of luck.

With regards,

closed 2024-09-23 by ALABSTM

vahidajalluian, 2024-09-25:
Sorry, I was away for some time, I have found that the reason arise from updating. If one create project purely by the newest case, that problem wont happen and all things work correctly, On the contrary, if the project is created by the past versions and then upgraded, it wont make the device descriptor code and hence, although the code is created by the latest version but it doesn't shown in device manager, here is a minimal project.
[Uploading USB_Test_Not_working.zip…]()
[Uploading USB_TEST_Working.zip…]()