STMicroelectronics/STM32CubeF4 #182: Uninitialized netif->name Causes Incorrect Debug Logs in LwIP Example with NETIF_DEBUG Enabled

bricle, 2024-11-16:
**Caution**

**Describe the set-up**
 * custom board.
 * STM32CubeIDE  with least version of CubeF4 packege.

**Describe the bug**
There is a bug in the LwIP example provided in the STM32CubeF4 software package. In the lwip.c file, the `MX_LWIP_Init` function calls 

```c
netif_add(&gnetif, &ipaddr, &netmask, &gw, NULL, &ethernetif_init, &ethernet_input).
```

Within `netif_add`, the `netif_set_addr` function is invoked, which subsequently calls `netif_do_set_netmask`. During the execution of `netif_do_set_netmask`, the following debug statement is executed (when `NETIF_DEBUG` is enabled):
```c
LWIP_DEBUGF(NETIF_DEBUG | LWIP_DBG_TRACE | LWIP_DBG_STATE, ("netif: netmask of interface %c%c set to %"U16_F".%"U16_F".%"U16_F".%"U16_F"\n",
    netif->name[0], netif->name[1],
    ip4_addr1_16(netif_ip4_netmask(netif)),
    ip4_addr2_16(netif_ip4_netmask(netif)),
    ip4_addr3_16(netif_ip4_netmask(netif)),
    ip4_addr4_16(netif_ip4_netmask(netif))));
```
However, at this point, `netif->name[0]` and `netif->name[1]` have not yet been initialized, as their initialization happens later in the `ethernetif_init` function.
```c
err_t ethernetif_init(struct netif *netif)
{
  LWIP_ASSERT("netif != NULL", (netif != NULL));

#if LWIP_NETIF_HOSTNAME
  /* Initialize interface hostname */
  netif->hostname = "lwip";
#endif /* LWIP_NETIF_HOSTNAME */

  /*
   * Initialize the snmp variables and counters inside the struct netif.
   * The last argument should be replaced with your link speed, in units
   * of bits per second.
   */
  // MIB2_INIT_NETIF(netif, snmp_ifType_ethernet_csmacd, LINK_SPEED_OF_YOUR_NETIF_IN_BPS);

  netif->name[0] = IFNAME0;
  netif->name[1] = IFNAME1;
...
}
```

You can see, `netif_set_addr` is called before  `init` function.

```c
#if LWIP_IPV4
  netif_set_addr(netif, ipaddr, netmask, gw);
#endif /* LWIP_IPV4 */

  /* call user specified initialization function for netif */
  if (init(netif) != ERR_OK) {
    return NULL;
  }
```

 As a result, the debug output for `netif->name` produces incorrect or undefined values.

**How To Reproduce**
1.  Application behavior: The application uses the default LwIP example generated by STM32CubeMX or provided in the CubeF4 software package.
2.  Suspected modules: LwIP middleware and netif implementation.
3.  Use case:
* Enable the NETIF_DEBUG option in lwipopts.h.
* Run the LwIP example on the provided board.
* Observe the debug logs where netif->name is used before it is initialized.
4.  Reproduction steps:
* Generate a project with LwIP middleware enabled using STM32CubeMX.
* Ensure NETIF_DEBUG is enabled.
* Build and run the project.
* Monitor the logs .

**Additional context**
Proposed Fix:
The issue can be resolved by ensuring that `netif->name` is initialized before calling `netif_add`.

**Screenshots**

ASEHSTM, 2024-11-21:
Hello @bricle 
Thank you for your report. All our LwIP examples function correctly, even with the call to netif_set_addr before the initialization of netif->name. Could you please provide more details on why this caused an issue for you?

With Regards,

ASEHSTM, 2025-03-11:
Hello @bricle,

Please allow me to close this thread as no activity.

With Regards,

closed 2025-03-11 by ASEHSTM