STMicroelectronics/STM32CubeF4 #183: Hello,

ostifede02, 2024-11-23:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.

**Describe the bug**
A clear and concise description of what the bug is.

**How To Reproduce**
1. Indicate the global behavior of your application project.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).

3. The use case that generates the problem.

4. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ostifede02, 2024-11-23:
I have encountered the following issue:
1. create new project from CubeMX and generated code with Toolchain/IDE: CMake

TOUNSTM, 2024-11-25:
Hello  @ostifede02,

The point you reported is actually related to the CubeMX generation problem and not to the firmware published in this repository. Unfortunately, we don't treat aspect related to CubeMX tool in our GitHub repositories. They are rather treated at the [STM32CubeMX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated page of the ST Community .

Since this issue is not directly related to the STM32Cube firmware but rather to our ecosystem, please allow me then to close this thread. Thank you for your comprehension.

With regards,

closed 2024-11-25 by TOUNSTM