STMicroelectronics/STM32CubeF4 #185: Issue when using BSP LCD driver for my board and ADC with DMA.

yummbee, 2025-02-14:
On the STM32F429I-Discovery1 board using CubeIDE, I'm encountering an issue where the ADC DMA interrupt is only triggered once when running in triple interleaved ADC mode.  The expected behavior is for the DMA interrupt to be called repeatedly in circular mode to continuously acquire ADC data.

This issue seems to potentially be related to LCD initialization using the BSP.  It's observed that when the LCD is initialized, the ADC DMA interrupt stops triggering after the initial call.  Without LCD initialization, the interrupt is repeatedly called as expected. 

Debugging Steps Already Taken:

Checked DMA Mode Configuration: Verified DMA is configured in Circular Mode in CubeIDE.
Checked ADC Continuous Conversion Mode: Confirmed ADC is set to Continuous Conversion Mode in CubeIDE.
Checked Interrupt Enable Configuration: Verified DMA Transfer Complete Interrupt (TCIE) is enabled in CubeIDE DMA settings and the corresponding DMA interrupt is enabled in NVIC.

I also see that if i first enable the ADC and DMA, the BSP_LCD_Init(); function does not execute. It tends to hang at different points but if i debug step through it i get no issues. 

I am using FreeRTOS and its corresponding interrupts. The freertos task can continue executing its just the DMA that ceases.

RJMSTM, 2025-02-18:
Hello @yummbee,

First thanks for your contribution, can you please try initializing the LCD before setting up the ADC and DMA. This way, any potential conflicts or configurations done by the `BSP_LCD_Init()` function won't interfere with the ADC setup.

Otherwise, would it be possible to share the project you used to reproduce the issue? This would allow for a better analysis of the problem.

 Best regards.

RJMSTM, 2025-06-17:
Hello,
Could you find the time to try our solution? We would greatly appreciate any feedback you may have.

yummbee, 2025-06-17:
Hi, my apologies, I completely missed this. I figured out the issue after a while. If I remember correctly, it turned out that the LCD used DMA stream 0 or 2, I cant fully remember, and my ADC also used the same DMA stream as configured by default in the IOC within STMCubeIDE. It took a while to find, and is very likely due to me missing this in the docs somewhere, but I eventually monitored the registers using the SFR live expressions and saw the channel is, at some point (and I do believe I saw it somewhere within the source code for the BSP but it's been a while) set to the same channel that my ADC DMA requests were being done over. This was very much due to me not understanding that the LCD uses DMA. 

Sorry for such a delayed response.