STMicroelectronics/STM32CubeF4 #186: Are the tagged versions supposed to include driver header/sources?

sylt, 2025-02-19:
**Describe the set-up**
* Go to https://github.com/STMicroelectronics/STM32CubeF4/tags
* Download the zip for v1.28.1 (latest as of writing this).
* Try: `unzip -l STM32CubeF4-1.28.1.zip | grep STM32F4xx_HAL_Driver`

As can be seen, it's just one entry, meaning the driver folder is empty with no headers nor sources. 

Is this intended? When reading the description of this project, I got the feeling it would include everything. 

The same issue exist in several ST device specific repos, so perhaps I'm misunderstanding something here.

sylt, 2025-02-19:
This goes for the various BSP folders as well -- are they supposed to be empty in the downloaded zip/tar.gz file?

It seems to be related to which paths in this repo which are subrepos.

bmcdonnell-fb, 2025-02-19:
Probably related to the fact that [they're using Git submodules](https://github.com/STMicroelectronics/STM32CubeF4/blob/master/README.md#how-to-use).

ALABSTM, 2025-02-21:
Hi @sylt,

Indeed, this is a **known limitation** when using the Git submodules. The better is to clone the repository using the `--recursive` option, as shown below.

```bash
git clone --recursive https://github.com/STMicroelectronics/STM32CubeF4.git
```

This limitation is indicated in the last note at the **bottom** of the [How to use](https://github.com/STMicroelectronics/STM32CubeF4?tab=readme-ov-file#how-to-use) section of the README.md. In case you prefer downloading the repository as a ZIP file, you have to **manually** collect and integrate the missing submodules.

With regards,

sylt, 2025-02-21:
Alright, I see! I was unaware of this github limitation, and from some research, it doesn't seem to be fixable using github actions either. That is, unless you switch to doing proper releases, as opposed to just tagging. Then you can create a job which clones the repo recursively.

So, consider that last part, but all in all, the question has been answered, so thank you! Closing this.