STMicroelectronics/STM32CubeF4 #189: Problem with LWIP 2.1.2 on STM32F407VET6

NguyenNgocAnh2610, 2025-03-27:
Hi, I'm having a problem with LWIP 2.1.2 on STM32F407VET6. Specifically, I am using LWIP 2.1.2 with the DP83848, generating code with CubeMX, and using Keil C5. My program doesn't run properly right after I upload the code to the board. When I run the debugger, I have to press "Run" (F5) three times for the program to run normally. I found that the cause is the MX_LWIP_Init(); line that CubeMX generates. If I comment out the MX_LWIP_Init() line, the program runs normally after uploading the code. Please help me fix this.

NguyenNgocAnh2610, 2025-03-28:
https://stackoverflow.com/questions/71908956/lwip-on-stm32f107-work-only-with-debugger-w-stmcubemx-and-keil. the problem i have is same as this case. i tried many ways but no success

ASEHSTM, 2025-04-02:
Hello @NguyenNgocAnh2610,

Could you please specify the versions of CubeMX and Keil that you are using?

With Regards,

ASEHSTM, 2025-04-09:
Hello @NguyenNgocAnh2610,

Any update regarding this thread?

With Regards,

NguyenNgocAnh2610, 2025-04-11:
Thank you @ASEHSTM . I solved this problem, but I encountered another problem. When I create a http webserver with stm32f407vet6 and dp83848, using HAL, LWIP. I configured everything according to the instructions here https://controllerstech.com/stm32-ethernet-6-http-webserver-simple/. When I load the program, I can ping from PC to stm32. But when I access the http webserver, there is no response, and my ping command is timeout. It seems there is a problem with the docubeide source code. I am using cubeide Version: 1.18.0, with CubeMX 6.14

NguyenNgocAnh2610, 2025-04-11:
Thank you. I solved this problem, but I encountered another problem. When I create a http webserver with stm32f407vet6 and dp83848, using HAL, LWIP. I configured everything according to the instructions here https://controllerstech.com/stm32-ethernet-6-http-webserver-simple/. When I load the program, I can ping from PC to stm32. But when I access the http webserver, there is no response, and my ping command is timeout. It seems there is a problem with the docubeide source code. I am using cubeide Version: 1.18.0, with CubeMX 6.14. Please help me fix this error! Thank you

ASEHSTM, 2025-04-15:
Hello @NguyenNgocAnh2610,

I'm glad to hear that you resolved the initial issue. 
Regarding your second problem, could you please open it in a separate issue for better traceability and management?

Thank you for your understanding.

With Regards,

ASEHSTM, 2025-04-18:
Hello @NguyenNgocAnh2610,

Thank you for opening a new issue. Please allow me to close this one now.

With Regards

closed 2025-04-18 by ASEHSTM