STMicroelectronics/STM32CubeF4 #191: Problem with CubeMX 6.14

NguyenNgocAnh2610, 2025-04-16:
Hello. I have a problem with CubeMX6.14. When I create a http web server with stm32f407vet6 and dp83848, using HAL, LWIP. I configured everything according to the instructions here https://controllerstech.com/stm32-ethernet-6-http-webserver-simple/. When I load the program, I can ping from PC to stm32. But when I access the http web server, there is no response and my ping command times out. I am using cubeide Version: 1.18.0, with CubeMX 6.14. When I try with CubeMX 6.3, everything works fine.
I hope that ST will check and fix the bug in the new update.

ASEHSTM, 2025-04-23:
Hello @NguyenNgocAnh2610,

Thank you for your report. After an initial analysis, CubeMX 6.14 seems to function correctly with other server applications in H5 and H7. However, I will conduct a more thorough analysis to try and reproduce your issue, and I will get back to you as soon as possible. 

In the meantime, could you please provide detailed information about your debug results, specifically indicating which part or function your code fails in?

Thank you.

With Regards.

ASEHSTM, 2025-05-13:
Hello @NguyenNgocAnh2610,

Any updates regarding this thread?

With Regards,

ASEHSTM, 2025-05-19:
Hello @NguyenNgocAnh2610,

Please allow me to close this issue due to inactivity.

If you need further assistance, feel free to open a new issue at any time.

With Regards,

closed 2025-05-19 by ASEHSTM