STMicroelectronics/STM32CubeF4 #195: Missing CMake files for CMSIS DSP

bmcdonnell-fb, 2025-06-25:
**Describe the set-up**
 * VSCode w/ CMake Tools extension

**Describe the bug**
Missing CMake files to configure/build CMSIS/DSP.
Also it's not documented.

**How To Reproduce**
1. Attempt to `add_subdirectory(Drivers/CMSIS/DSP/Source)` in a `CMakeLists.txt`.
2. You will find that you're missing at least the following files, available from [CMSIS_5 v5.9.0 (CMSIS/DSP/)](https://github.com/ARM-software/CMSIS_5/tree/5.9.0/CMSIS/DSP):
  - Toolchain/*.cmake
  - configCore.cmake
  - configLib.cmake
3. You'll also need to `set()` some variables that they consume, e.g. `PROJECTNAME` and `ROOT`.

**Additional context**
[How to integrate CMSIS-DSP libraries on a CMake STM32 project](https://community.st.com/t5/stm32cubemx-mcus/how-to-integrate-cmsis-dsp-libraries-on-a-cmake-stm32-project/td-p/812804)

bmcdonnell-fb, 2025-06-25:
Alternatively, jump ahead to using [CMSIS-DSP](https://github.com/ARM-software/CMSIS-DSP).

TOUNSTM, 2025-07-02:
ST Internal Reference: 213281

bmcdonnell-fb, 2025-07-25:
@PLFSTM thanks for the follow-up.

I believe this is ST's mistake, not ARM's.

To clarify: upstream `CMSIS/DSP` files can be found in at least two places:
1. [CMSIS_5 v5.9.0 (CMSIS/DSP/)](https://github.com/ARM-software/CMSIS_5/tree/5.9.0/CMSIS/DSP) (the `DSP` subdir of the CMSIS_5 repo, linked in OP here) - old location
2. The [CMSIS-DSP repo](https://github.com/ARM-software/CMSIS-DSP) - new location - associated with [CMSIS_6](https://github.com/ARM-software/CMSIS_6)?

AFAICT, so far, ST has been taking the `CMSIS/DSP` files from [CMSIS_5 v5.9.0](https://github.com/ARM-software/CMSIS_5/tree/5.9.0/CMSIS/DSP) (1), _not_ [the CMSIS-DSP repo](https://github.com/ARM-software/CMSIS-DSP) (2), where you posted https://github.com/ARM-software/CMSIS-DSP/issues/260.

As shown in the following diff summary of `STM32CubeF4/Drivers/CMSIS/` from [this repo, v1.28.2](https://github.com/STMicroelectronics/STM32CubeF4/tree/v1.28.2) vs [CMSIS_5 v5.9.0 (CMSIS/DSP/)](https://github.com/ARM-software/CMSIS_5/tree/5.9.0/CMSIS/DSP), this repo is missing the upstream files noted in my OP:

[image]

[image]

I believe ST should copy at least the missing files I noted from [CMSIS_5 v5.9.0 (CMSIS/DSP/)](https://github.com/ARM-software/CMSIS_5/tree/5.9.0/CMSIS/DSP) to this repo.

PLFSTM, 2025-07-30:
@bmcdonnell-fb Thank for your deep analysis.

ST imported CMSIS/DSP files from the official pack [ARM.CMSIS.5.9.0.pack](https://github.com/ARM-software/CMSIS_5/releases/download/5.9.0/ARM.CMSIS.5.9.0.pack) which sounds not to take care of CMake build back in 2022.
A solution may rather consist ST to move to latest CMSIS Base Software Library pack v6.2.0 and CMSIS-DSP Software Library pack v1.16.2 so to benefit from latest updates and fixes.
Hope you managed to complete your build by picking up the missing files from [CMSIS_5 v5.9.0 (CMSIS/DSP/)](https://github.com/ARM-software/CMSIS_5/tree/5.9.0/CMSIS/DSP)

bmcdonnell-fb, 2025-08-03:
> ST imported CMSIS/DSP files from the official pack [ARM.CMSIS.5.9.0.pack](https://github.com/ARM-software/CMSIS_5/releases/download/5.9.0/ARM.CMSIS.5.9.0.pack) which sounds not to take care of CMake build back in 2022.

Your link excludes the whole `DSP/Toolchain/` subdir. (Maybe a previous version had it?) But why take your files from the pack instead of the upstream repo anyway?

> A solution may rather consist ST to move to latest CMSIS Base Software Library pack v6.2.0 and CMSIS-DSP Software Library pack v1.16.2 so to benefit from latest updates and fixes.

On a quick check, it seems the [CMSIS-DSP repo](https://github.com/ARM-software/CMSIS-DSP) is missing some of the same CMake files, but may not need them. I haven't tried to build from that.

> Hope you managed to complete your build by picking up the missing files from [CMSIS_5 v5.9.0 (CMSIS/DSP/)](https://github.com/ARM-software/CMSIS_5/tree/5.9.0/CMSIS/DSP)

I did, thanks. Just trying to push fixes upstream.

closed 2025-08-29 by TOUNSTM